rwaldron/johnny-five #567: Get rid of toNormal() in Gyro class

BrianGenisio, 2015-01-04:
The `Gyro` class currently uses an arbitrary, unit-less measure for X, Y, and Z, supported by the `toNormal()` function and centers at 127.

0. Change it so that X, Y, and Z use `toDegreesPerSecond()` and center at 0 (no rotation).  
0. Get rid of `rate` because it is now the same as X, Y, and Z
0. Change anything currently using `rate` internally to use X, Y, and Z

reconbot, 2016-08-24:
@BrianGenisio is this still worth doing? Up for the challenge, we now have a few controllers using it.

rwaldron, 2016-08-24:
@reconbot only because the class expects it there. 

We should do this: 

- Remove toNormal, put whatever functionality each is doing into the `toDegreesPerSecond` method
- Officially move the values expected at `this.rate.{x, y, z}` to `this.{x, y, z}` 
- Document it
- Change all the examples
- Maintain a branch until 0.11.0 and merge it.

closed 2019-01-10 by stale[bot]