rwaldron/johnny-five #569: Standard Event object

rwaldron, 2015-01-05:
### Proposal: All events will emit a standard Event object.

```js
function Event(opts) {
  this.timestamp = Date.now();
  this.type = opts.type || "data";
  this.data = opts.data || null;
}
```

```js
// Example
this.emit("data", new Event({ data: adc }));
```

**This is a breaking change**

rwaldron, 2015-01-05:
It occurs to me that we can limit how much this breaks by making this a standard _second_ argument.

dtex, 2015-01-05:
What would the first argument be if not the faux error?

rwaldron, 2015-01-05:
Whatever was previously the second argument :\

scottgonzalez, 2015-04-28:
If you're changing the signatures with #561 anyway, I'd say you should just go all the way and do the standardization to a single argument.

dtex, 2018-08-21:
Someone recently agreed to look at this issue so I am commenting to keep it fresh.

dtex, 2019-08-16:
Not ready to close this one Stale Bot.

HipsterBrown, 2019-08-17:
@ajfisher was showing me his audit of all the callbacks and event objects in the codebase. I wonder how we could share that somewhere.

closed 2020-08-19 by stale[bot]