rwaldron/johnny-five #572: Aliases in johnny-five.js are not documented and possibly in the wrong place

dtex, 2015-01-06:
We have this in johnny-five.lib
````
// Short-handing, Aliases
module.exports.Boards = module.exports.Board.Array;
module.exports.ESCs = module.exports.ESC.Array;
module.exports.Leds = module.exports.Led.Array;
module.exports.Servos = module.exports.Servo.Array;
````

1. We need to make sure that these are all documented
2. We should discuss moving these to their respective classes so they are easier to find

OR

Should these even exist? 

Removing them would be a breaking change but 0.9 might be a good time to remove aliases and stuff that exist for no other reason than backwards compatibility.

rwaldron, 2015-01-06:
> We need to make sure that these are all documented

Yes. 

> We should discuss moving these to their respective classes so they are easier to find

These can't move, all class files export a single "default" binding, so cannot export both the class definition binding and the alias. 

> Should these even exist?

Why not? I'd rather people use these names vs. the long form name.

dtex, 2015-12-14:
I've documented Servos on a separate page of the wiki. I dropped it at the bottom of the Servo page at first but that page is already soooo long. I added a link on Servo to get to Servos, but I'm not sure how that will work on johnny-five.io

https://github.com/rwaldron/johnny-five/wiki/Servos

If that works, I'll add in docs for LEDs and ESCs real quick so we can close this.

rwaldron, 2015-12-14:
Awesome! I've added it here: https://github.com/rwaldron/johnny-five/wiki and it will go up to the site on the next deployment. Things I'd like to do: 

- Have diagrams that show > 1 servo connected
- An example that shows controlling a directly connected servo and a PCA9685 servo in the same array

dtex, 2015-12-14:
Okay, will do. Also since we are adding a direct link at the top level I'm going to add a note at the top of the Servos wiki page for users who are just looking to control one servo.

rwaldron, 2015-12-15:
Sounds good

dtex, 2015-12-17:
All are documented and landed

closed 2015-12-17 by dtex