rwaldron/johnny-five #577: Support for HT1632C

jeffreynolte, 2015-01-14:
Hi! 

I wanted to see if it is possible to get support for the HT16K33 in Johnny 5.
Here is the product on http://www.adafruit.com/product/555
Datasheet: http://www.adafruit.com/datasheets/ht1632cv120.pdf
Here is the library from Adafruit: http://www.adafruit.com/datasheets/ht1632cv120.pdf

julianduque, 2015-01-14:
HT16K33 is supported, take a look at: https://github.com/rwaldron/johnny-five/wiki/Led.Matrix

jeffreynolte, 2015-01-14:
Thanks @julianduque. Closing.

closed 2015-01-14 by jeffreynolte

rwaldron, 2015-01-14:
@jeffreynolte I just realized that the example program wasn't linked to from the readme: https://github.com/rwaldron/johnny-five/blob/master/docs/led-matrix-HT16K33.md

jeffreynolte, 2015-01-14:
Thanks @rwaldron !

frxnz, 2015-01-14:
This kind of brings up something I was thinking about last night, as I was trying to figure out if J5 supported the Adafruit PWM driver. Eventually I found my answer by searching closed issues.

Right now it's hard to figure out if J5 supports something. As we continue adding support for devices it's going to get harder. It would be nice if there was a searchable webpage or something.