rwaldron/johnny-five #579: Accelerometer and Gyro events are not consistent

BrianGenisio, 2015-01-15:
Now that we are working on some breaking changes in 0.9.0, it is probably worth noting that the `data` event on `Accelerometer` emits the raw data, where `change` emits the exposed values. This felt odd to me when I got to it, but I didn't want to break the existing interface.  In contrast, `Gyro`  sends the exposed values for `data` and `change` events.  

It feels like we should be consistent with this.  Should it work like `Accelerometer`, where `data` is raw and `change` is shaped?  Or like `Gyro` where both events emit the same "shaped" data?

dtex, 2015-01-15:
I'm really glad you opened this. When I first played with IMU I had been listening for change and had seen IMU was working. Now that I'm working on imp-io I2C support my test code uses data events which yields different results. I have been banging my head on the keyboard since yesterday not realizing that the difference was a result of listening to different events between the Imp and Aruino code.

Clearly I'm in the "both should be shaped" camp :-)

IMHO, both should be shaped.

BrianGenisio, 2015-01-15:
At the same time, the "raw" values can be useful.  For example, if I am a user, trying to integrate an unknown analog accelerometer in my project, I should be able to do so by setting the `zeroV` and `sensitivity` properties.  I can do this by inspection if I have the raw values.

Now that we have a good "controller" and "translation" pattern, I wonder if it is worth emitting a new event?  A "raw" event can be something that gets emitted whenever "data" is emitted.  It wouldn't be used too often, but it would enable this use case, as well as diagnostic/debug/development use cases.

closed 2019-01-10 by stale[bot]