rwaldron/johnny-five #580: What does Gyro angle mean?

BrianGenisio, 2015-01-15:
Since we are making breaking API changes in 0.9.0, it seems to me that this can use some discussion as well.

Similar to the ["normal"](https://github.com/rwaldron/johnny-five/issues/567) issue, the `angle` property on `Gyro` seems wrong.  The `angle` property is currently an accumulator of the `rate / 100`.  "Rate" is well defined: degrees/second.  Knowing this, `state.x.angle += this.rate.x / 100;` feels arbitrary.  In addition to this, the `rate` value is only reported at that moment in the sampling rate, so accumulating this value is very error prone.  It constantly drifts and becomes (seemingly) meaningless over time.

Honestly, I don't know what `angle` is trying to achieve. It is my understanding (perhaps incorrect) that gyros can't give you an angle, only the change in rotation.  If you want an angle, you need to ask the accelerometer which gives you orientation, but not rotation.  

Do we know if anyone is using `angle` for anything?  If so, what does `angle` achieve?  If not, what should it achieve?  Or should we yank it?  Am I just confused?

rwaldron, 2015-01-17:
IIRC this was actually angular velocity and named badly, probably incorrect math as well. Let's ger rid of it

canda, 2015-09-29:
Isn't it possible to measure the actual angle/position by adding the rates multiplied by time elapsed for each sample over time? Is it precise enough?

closed 2018-08-28 by stale[bot]