rwaldron/johnny-five #585: Replace calls to console.log with Board.log

dtex, 2015-01-19:
Board.log gives color coding, timestamps, and other nifty things. More importantly adds a single point to re-route all log data making it easier to use on headless or remote host servers.

divanvisagie, 2015-01-25:
has anyone done this yet? if not Ill grab it.

ColtraneNadler, 2015-10-06:
has this been completed yet? if not I would take this on in spirit of Hacktoberfest :)

sevastos, 2015-10-28:
Should `console.log` of deprecation notices be turned to Board.log as well?
If so, during tests there won't be any messages printed for using a deprecated function.
For example [LedControl.setLed](https://github.com/rwaldron/johnny-five/blob/master/lib/led/ledcontrol.js#L194-L195).

reconbot, 2016-05-25:
@sevastos yes

This is still an open issue for some of the codebase.

reconbot, 2018-08-21:
This is no longer an issue afact

closed 2018-08-21 by reconbot