rwaldron/johnny-five #587: Simultaneous button events get dropped

dtex, 2015-01-23:
If two buttons are pressed or released simultaneously, one of the two button's events will not fire. It is easily reproduced with imp-io because we send pin states in batches but it happens with Firmata on the Uno as well... It's just harder to reproduce. With my ninja like reflexes I'm able to make it happen about 10% of the time on the Uno.

The code to reproduce is pretty straightforward
````
var five = require("johnny-five");

var board = new five.Board();

board.on("ready", function() {

  var b1 = new five.Button(3);
  var b2 = new five.Button(11);
  b1.state = b2.state = "off";

  b1.on("press", function() {
    b1.state = "on";
    report("b1");
  });

  b1.on("up", function() {
    b1.state = "off";
    report("b1");
  });

  b2.on("press", function() {
    b2.state = "on";
    report("b2");
  });

  b2.on("up", function() {
    b2.state = "off";
    report("b2");
  });

  report = function(trigger) {
    console.log(trigger, b1.state, b2.state);
  }
});
````

Then just press and/or release the two buttons at EXACTLY the same time.

dtex, 2015-01-23:
Probably related to #152

soundanalogous, 2015-01-23:
I may be wrong here, but since [trigger](https://github.com/rwaldron/johnny-five/blob/master/lib/button.js#L16-L20) is defined outside of the constructor it would be called for each instance of Button. In that case if you press two or more buttons within the debounce threshold the second press will debounce the first. Each Button needs to handle it's own debounce function.

soundanalogous, 2015-01-23:
Try moving [these lines](https://github.com/rwaldron/johnny-five/blob/master/lib/button.js#L16-L20) to the Button constructor (so `this.trigger...`) and see if that makes a difference.

rwaldron, 2015-01-23:
Before I got half way through the OP I was already thinking what @soundanalogous was thinking :)

PR #588 (2015-01-23): Move debounce function to constructor
Fixes #587 

Prevents “cross-debouncing” of events across button instances.

**Don't merge yet.** - I don't have hardware with me to test.
closed 2015-01-23 by rwaldron