rwaldron/johnny-five #606: support for RTC (real-time clock)

boneskull, 2015-01-29:
A popular (I think?) RTC chip is the DS1307.  It's included in many (several? some?) data logger shields.

I'm working on support for interfacing w/ the DS1307 directly, but this is my first encounter with I2C, and I am having trouble translating the Arduino library (`Wire` calls) to the J5 equivalent.  Any general tips would be helpful here, since the docs seem kind of slim in this area unless I missed something.

I *think* the implementation should be a `Controller` in a new `RTC` (or `Clock` perhaps) class, maybe even the `DEFAULT` one.  I imagine tests, docs, and a trivial example are expected.

Is there anything else I need to know?

dtex, 2015-01-29:
Firmata has a handful of I2C methods you can call so you shouldn't have to worry about the wire stuff. Check out the methods here:

https://github.com/jgautier/firmata#i2c

boneskull, 2015-01-29:
@dtex Yes, but I'm having trouble figuring out the library for this thing that uses `Wire`; the mapping between what method *there* corresponds to what method *here*.

dtex, 2015-01-29:
It might be easier to get the I2C info straight from the [DS1307 docs](http://datasheets.maximintegrated.com/en/ds/DS1307.pdf)

boneskull, 2015-01-29:
@dtex I'm looking at that, but as I said, I have not encountered I2C before, so I'm just kind of flailing.

rwaldron, 2015-01-29:
Can you post a link to the component you're working with? (so I can follow along :D)

rwaldron, 2015-01-29:
First of all, I like the idea of being able to interact with a real time clock, but I'm wondering what advantage this has over system time in ms available to JS via `Date.now()`. RTC components are generally used with platforms that have no "built-in" notion of time (beyond the "time since this processor started"). Johnny-Five runs in an environment that already has access to time. 

To clarify, I didn't bring this up to discourage you, I just want to talk about these aspects and get a better idea of use cases and so forth.

boneskull, 2015-01-29:
It's [this](http://www.maximintegrated.com/en/products/digital/real-time-clocks/DS1307.html).

It's included in [this shield](https://www.adafruit.com/products/1141), which I have a rough Chinese clone of.

boneskull, 2015-01-29:
@rwaldron The clock is used by the data logger to record the time when logging.  Given what you just wrote, I'm having a hard time envisioning a use case outside of this.  Perhaps it's better just to go with #607 instead of this.

boneskull, 2015-01-29:
Sh*tcanning this for reasons discussed in #607.

closed 2015-01-29 by boneskull

rwaldron, 2015-01-29:
So, I think it might be fun for you to do this anyway, just to get an understanding for how the I2C API and semantics work :) I looked up some super raw examples and discovered that reading these devices is pretty straight forward (cursory look, I might be wrong). Either way, in the end you'll have learned how to use the I2C functionality :)

rwaldron, 2015-01-29:
I will get this one https://learn.adafruit.com/ds1307-real-time-clock-breakout-board-kit/overview

rwaldron, 2015-01-29:
j/k, I got this one: http://www.amazon.com/Real-Time-Clock-Module-DS1307/dp/B00BLC3C2Y

aabm00, 2017-02-08:
Hi

In my project I am running Johnny-five in an Arduino Yun, and have to send data to a server every 30 seconds with the real date and time. It's a real-time application controlling machines and an accurate date-time is important.
The problem with Yun is that it hasn't a real time clock, so it cannot track the time when it is off. While booting, the Linux side makes some educated guesses of the current date/time by checking the modification time stamps on key files, but it doesn't really know the time until it has an Internet connection and can query an NTP server (sometimes it doesn't work), or until you manually set the date and time . 
The most reliable method to make sure the Yun has the correct local time is to add a battery backed up RTC chip to it. 

I have had problems with YUN sending data with date of 20/07/2015 so it would be fine to have a J5 implementation of [DS1307](https://www.adafruit.com/product/3296) and/or [DS3231 ](https://learn.adafruit.com/adafruit-ds3231-precision-rtc-breakout/overview) to this use cases when the date is important and the board with connection to internet doesn't have a real time RTC clock.

Thanks