rwaldron/johnny-five #607: support for data logger shields

boneskull, 2015-01-29:
I haven't investigated the actual data logging interface of my data logger shield, but once I do, I imagine an example or class would be in order.

rwaldron, 2015-01-29:
Possibly, if the logged data can be accessed via I2C (would have to encode every character and be limited to the 32 byte read limit imposed by `Wire` library). 

I'm not sure I understand what a data logger shield can offer to a platform that runs in a full-featured environment with uninhibited access to the file system.

boneskull, 2015-01-29:
@rwaldron Again, this is a good point.  I can't think of anything right now.  Given that *in order to use JavaScript with an Arduino* I must be tethered to a computer, and if I am, what's the point of the shield?  

I don't know.  It was like, $4.  I just bought it.  You know how it is.

closed 2015-01-29 by boneskull

rwaldron, 2015-01-29:
> Given that in order to use JavaScript with an Arduino I must be tethered to a computer, 

Not necessarily, eg. Intel Galileo, Edison, Linino One, Arduino Yun, pcDuino (soon), 

> and if I am, what's the point of the shield?

...But yeah, this still holds :)

> I don't know. It was like, $4. I just bought it. You know how it is.

I sure do :)