rwaldron/johnny-five #608: Functionality for pre-loading characters for Led.Matrix

jakobengdahl, 2015-01-29:
I'm building a johnny-five based robot for my son with a lot of buttons and leds etc. I'm using a number of led matrices to draw a heart and a mouth and when pressing certain buttons - the robot should "sing" and flash RGB-lights. To animate the heart and the mouth I make use of the .draw-function which send the next characters to the matrices but this process slows down the other communication between the program and the arduino so I'm not able to get a nice flow for the RGB-light animations. 

I've uploaded a video here to try to explain the problem. http://youtu.be/ZCq9tkbS0Hw 

At the moment I've setup another arduino to control the mouth to enable the robot to "sing" without interfering with the RGB-eyes (to show what I'm aiming for) but the heart is controlled by the same program as the RGB-eyes and turning the heart animation on and off shows the problem. I'm also using ajfisher's node-pixel (https://github.com/ajfisher/node-pixel) at the moment but it looks like he is making progress in integrating the needed functionality directly into standard-firmata. (https://github.com/ajfisher/node-pixel)
 
Would it be possible to add a function for pre-loading a character using a new function and then use another function for telling the arduino to draw a specific pre-loaded character to minimize the communication between the program and the arduino when animating between a few characters.

I guess firmata might be a limitation for creating this functionality and if this is the case I'll just stick with using one or two separate arduinos for the led matrices but it would be awesome if this functionality could be added.

dtex, 2015-01-29:
I can't help with the issue, but I took a look at the video and I gotta' say your are probably the coolest dad ever.

rwaldron, 2015-01-29:
There's a lot going on here and I'm having trouble boiling it down to a reduced test/use case. The request is too scenario-specific itself, but I wonder what can be done to address the issue in a more generalized way. It would be helpful if you could provide the code and hopefully even a fritzing breadboard diagram so I could recreate—thanks! :)

jakobengdahl, 2015-01-30:
dtex: Thank you! I hope he will like it!

rwaldron: You're right - there are a lot of things in the project that most likely isn't connected to this issue. I think the current code on the application side covers to much and doing a fritzing diagram over the whole thing might be overkill. Instead I'll setup a new small project with only the rgb-leds and the matrices and I'll update this issue with information and code for this so it'll be easier to continue with this information. I'll get back to you with information about this.

rwaldron, 2015-01-30:
@jakobengdahl awesome, that will be very helpful :)

jakobengdahl, 2015-02-01:
Okay @rwaldron, I've made a much more simple case to illustrate my problem. This time using standardfirmata and just a plain led to avoid other possible causes for the problem.

I haven't used fritzing before and couldn't find the MAX7219 which I'm using but the connections are very simple so I think you understand how it's connected.

[image]

https://www.youtube.com/watch?v=empPIwM0Jc4

This video shows a basic blinking led (set to 100ms). When the matrix animation isn't active, the led flashes in a steady pace. After 5000ms the "animation" is started which causes the flashing led to "lag". After 10000ms the "animation" is stopped and the led goes back to a steady 100ms pace.

The rgb-strip that I have connected to the robot project behaves in the same way as the led in this example and my initial thought is that this isn't a bug or anything - it's just that this type of solution hits the limits of how much data that can be communicated between the application and the firmata before the pace of the animation etc starts to lag.

I also tried increasing the baud rate and could reach a somewhat better result when using 280000 as baud rate but the issue remains and adding more matrices and therefore sending more characters causes the same problem. I couldn't get the matrix to act reliable over 280000.  

Since the draw-function of the led matrix communicates the entire "character" that should be shown, a lot of data has to be passed to the arduino. My question is if it is possible to add functionality for pre-loading characters using a new function and have them buffered on the arduino side. When the characters are loaded to the arduino, animation could be preformed by calling another function which tells the arduino to show a specific pre-loaded character. This way the whole characters of the animation would not have to be transfered for each frame of the animation. 

Once again, I realize that there might be limitations here but I wanted to check with you if you think it would be possible.

Here is the code I used for this example:

var SerialPort = require("serialport").SerialPort;
var five = require("johnny-five");
var board = new five.Board({
  port: new SerialPort("/dev/ttyUSB0", {
    baudrate: 280000
  })
});

var matrix;
var tHeart=false;
var heartbig = [
      "01100110",
      "10011001",
      "10000001",
      "10000001",
      "01000010",
      "00100100",
      "00011000",
      "00000000"
    ];

    var heartsmall = [
      "00000000",
      "00100100",
      "01011010",
      "01000010",
      "00100100",
      "00011000",
      "00000000",
      "00000000"
    ];

board.on("ready", function() {
  matrix = new five.Led.Matrix({
    pins: {
      data: 2,
      clock: 3,
      cs: 4
    }, 
    devices: 1
  });
  matrix.on();
  (new five.Led(13)).blink(100);
  
  setTimeout(drawMatrix, 5000);
  setTimeout(function (){drawMatrix=null}, 10000)
});

function drawMatrix(){
  tHeart= !tHeart;      
  matrix.draw((tHeart)?heartbig:heartsmall);    
  setTimeout(drawMatrix, 300);
}

rwaldron, 2015-02-02:
(I edited the post to format the code)

rwaldron, 2015-02-19:
Just wanted to check in here to let you know that I haven't forgotten about this, I've just been swamped.

closed 2019-01-10 by stale[bot]