rwaldron/johnny-five #609: Animation not working with Arduino Starter Kit servo

cescopag, 2015-01-30:
I have problems with animation with the servo included in the Arduino Starter Kit.
If i try the servo by itself ( i.e. servo.sweep() ) it works fine, but if I try to use animation it doesn't move.

Thanks

dtex, 2015-01-30:
I'll try to help. Can you post your code?

cescopag, 2015-01-31:
Thank you dtex,
this code works:
```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var servo = new five.Servo(10);
  servo.sweep();
});
```

But this (taken from the documentation) doesn't:
```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var servo = new five.Servo(10);
  var animation = new five.Animation(servo);

  // Create an animation segment object
  animation.enqueue({
    duration: 2000,
    cuePoints: [0, 0.25, 0.5, 0.75, 1.0],
    keyFrames: [ 0, 135, 45, 180, 0]
  });
});
```

it got stuck to zero position.

I'm using Arduino Uno with the servo included in the starter kit:
http://farm9.staticflickr.com/8052/8109445971_202e309305_c.jpg

dtex, 2015-01-31:
That code is from an example isn't it? It's wrong. Short answer, this will work:

````
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var servo = new five.Servo(10);
  var animation = new five.Animation(servo);

  // Create an animation segment object
  animation.enqueue({
    duration: 2000,
    cuePoints: [0, 0.25, 0.5, 0.75, 1.0],
    keyFrames: [ {degrees: 0}, {degrees: 135}, {degrees:45}, {degrees:180},{degrees: 0} ]
  });
});
````
I'll go and check the docs now.

dtex, 2015-01-31:
I found the erroneous examples in the wiki and fixed them. I'm going to spend some time checking everything else in Animation and make sure it's all up to date.

cescopag, 2015-01-31:
Thank you very much! it works perfectly.

closed 2015-02-02 by dtex