rwaldron/johnny-five #611: Intel Edison: Segmentation fault when requiring johnny-five

mpbod, 2015-02-05:
I seem to be getting Segmentation fault on the Intel Edison
`
> five = require("johnny-five")
PID 384 received SIGSEGV for address: 0x200c
/node_app_slot/node_modules/segfault-handler/build/Release/segfault-handler.node(+0xcbd)[0xb7732cbd]
linux-gate.so.1(__kernel_rt_sigreturn+0x0)[0xb773640c]
node(_ZN2v88internal13GlobalHandles6CreateEPNS0_6ObjectE+0x40)[0x81cb6c0]
node(_ZN2v88internal19Runtime_ArrayConcatENS0_9ArgumentsEPNS0_7IsolateE+0x12ba)[0x830787a]
[0x29c0a236]
[0x29c5a0f0]
[0x29c0fe19]
[0x29c0a0aa]
node[0x81a8723]
node(_ZN2v88internal9Execution4CallENS0_6HandleINS0_6ObjectEEES4_iPS4_Pbb+0xc6)[0x81a9dc6]
node[0x817b28b]
node[0x817b760]
[0x29c0a236]
[0x29c58f56]
[0x3e723562]
[0x29c45f37]
[0x29c0f7a1]
[0x29c49ec0]
[0x29c49da4]
[0x3e73690e]
[0x29c141a5]
[0x29c47057]
[0x3e724d59]
[0x3e7235ff]
[0x29c45f37]
[0x29c0f7a1]
[0x29c49ec0]
[0x29c49da4]
[0x3e736449]
[0x29c0fe19]
[0x29c0a0aa]
node[0x81a8723]
`

rwaldron, 2015-02-05:
Thanks for the report. I need a lot more information than this ;)

1. With which Edison breakout? Mini or Arduino?
2. What version of Johnny-Five?
3. What version of Galileo-IO/Edison-IO?
4. Please provide the exact code that caused this
5. What is the output of this: `node -e 'console.log(require("mraa").getVersion());'`

Thanks!

rwaldron, 2015-02-05:
Also—were you trying to include Johnny-Five from an interactive node repl? This _will_ _not_ work.

mpbod, 2015-02-05:
Hi, thank you for the prompt reply.

1. Arduino
2 & 3. 
`
johnny-five@0.8.40 node_modules/johnny-five                                                                                                                                       
├── ease-component@1.0.0                                                                                                                                                          
├── descriptor@0.1.0                                                                                                                                                              
├── temporal@0.4.0                                                                                                                                                                
├── color-convert@0.5.2                                                                                                                                                           
├── colors@1.0.3                                                                                                                                                                  
├── es6-shim@0.24.0                                                                                                                                                               
├── nanotimer@0.3.1                                                                                                                                                               
├── firmata@0.4.0 (object-assign@1.0.0, browser-serialport@2.0.1)                                                                                                                 
├── array-includes@1.0.5 (es-abstract@1.0.1, define-properties@1.0.1)                                                                                                             
├── lodash@3.1.0                                                                                                                                                                  
├── galileo-io@0.7.0 (graceful-fs@2.0.3, es6-promise@2.0.1, remapped@0.2.1)                                                                                                       
└── serialport@1.4.10 (bindings@1.2.1, sf@0.1.7, async@0.9.0, nan@1.3.0, optimist@0.6.1)   
`

4.
`
//var SegfaultHandler = require('segfault-handler');
//SegfaultHandler.registerHandler();
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
    var led = new five.Led(13);
    led.strobe();
});
`
5. v0.5.4-110-g459ecc0

Running from Intel XDK IoT and Repl yields same results.

Thanks

yangjae, 2015-02-06:
I have the same issue.

rwaldron, 2015-02-06:
I worked on this for hours last night and I feel like I got nowhere. Take a look at this bug: https://github.com/rwaldron/galileo-io/issues/24 from less than a month ago—the OP and I confirmed that all three test cases (and more) worked as expected. 

Here's what I know: 

- Galileo-IO/Edison-IO work fine in standalone scripts
- Johnny-Five latest version (0.8.40) will crash AND the version from 2 weeks ago (0.8.36)
  - This is maddening because how is that even possible, if the test cases were independently confirmed??
- Something is going on with initialization.

This is as much as I know so far

rwaldron, 2015-02-06:
Rolled back to: 

- mraa@0.5.4
- galileo-io@0.3.3
- johnny-five@0.8.22

And the this example: 

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  console.log("ready");
  var led = new five.Led(11);
  led.blink();

});
```

(Edison-IO is just an alias to Galileo-IO)

rwaldron, 2015-02-06:
This: 

- mraa@0.5.4
- galileo-io@0.5.4
- johnny-five@0.8.22

Result: 

- No crash
- Blinking light

Success.

Next: 

- Why?

rwaldron, 2015-02-06:
Forget it, that's not quite right either.

mpbod, 2015-02-06:
Almost there! Thank you for your help @rwaldron!

rwaldron, 2015-02-06:
Update: 

- johnny-five@0.8.41-dev-002
- galileo-io@0.5.4
- mraa@0.5.4

Result: 

- no crash
- blinking light
- can control from repl

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  console.log("ready");
  var led = new five.Led(11);
  led.blink();

  this.repl.inject({
    led: led
  })
});
```

rwaldron, 2015-02-06:
Update: 

- johnny-five@0.8.41-dev-002
- galileo-io@0.5.4
- mraa@0.5.4-110-g459ecc0

```diff
- - mraa@0.5.4
+ - mraa@0.5.4-110-g459ecc0
```

Result: 

- no crash
- blinking light
- can control from repl

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  console.log("ready");
  var led = new five.Led(11);
  led.blink();

  this.repl.inject({
    led: led
  })
});
```

rwaldron, 2015-02-06:
Update: 

- johnny-five@0.8.41-dev-002
- galileo-io@0.5.4
- mraa@0.5.4-110-g459ecc0

```diff
- - galileo-io@0.5.4
+ - galileo-io@0.5.6
```

Result: 

- no crash
- blinking light
- can control from repl

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  console.log("ready");
  var led = new five.Led(11);
  led.blink();

  this.repl.inject({
    led: led
  })
});
```

fivdi, 2015-02-06:
The last diff doesn't look quite right. Should the following not be directly above it?

 * johnny-five@0.8.41-dev-002
 * galileo-io@0.5.6
 * mraa@0.5.4-110-g459ecc0

rwaldron, 2015-02-06:
Update: 

- johnny-five@0.8.41-dev-002
- galileo-io@0.6.4
- mraa@0.5.4-110-g459ecc0

```diff
- - galileo-io@0.5.6
+ - galileo-io@0.6.4
```

Result: 

- no crash
- blinking light
- can control from repl

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  console.log("ready");
  var led = new five.Led(11);
  led.blink();

  this.repl.inject({
    led: led
  })
});
```

rwaldron, 2015-02-06:
@fivdi yes, that was some copy/pasta, it's fixed.

rwaldron, 2015-02-06:
Update: 

- johnny-five@0.8.41
- galileo-io@0.6.4
- mraa@0.5.4-110-g459ecc0

```diff
- - johnny-five@0.8.41-dev-002
+ - johnny-five@0.8.41
```

Result: 

- no crash
- blinking light
- can control from repl

```js
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  console.log("ready");
  var led = new five.Led(11);
  led.blink();

  this.repl.inject({
    led: led
  })
});
```

mpbod, 2015-02-07:
Everything is now working, please use the following versions of the modules:
- johnny-five@0.8.41
- galileo-io@0.8.0
- mraa@0.5.4-110-g459ecc0

Please should require galileo-io / edison-io before johnny-five, example below:
```javascript
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
  console.log("ready");
  var led = new five.Led(13);
  led.blink();

  this.repl.inject({
    led: led
  })
});
```

**If anything breaks, first try to "hard" power cycle** the Edison by removing all power (including USB) and connecting them again.

rwaldron, 2015-02-07:
:clap:

closed 2015-02-07 by rwaldron