rwaldron/johnny-five #612: Intel Edison: Creating HT16K33 Led Matrix causing crash.

nickalto, 2015-02-06:
The following main.js is causing a crash on my edison: 

```
var mraa = require("mraa");
var five = require("johnny-five");

console.log("MRAA Version: " + mraa.getVersion()); 

var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {
	
   var wink = [
    "0000000000000000",
    "0011110000000000",
    "0100001000000000",
    "1001100111111111",
    "1001100111111111",
    "0100001000000000",
    "0011110000000000",
    "0000000000000000",
  ];

  var matrix = new five.Led.Matrix({
    addresses: [0x70],
    controller: "HT16K33",
    dims: "8x16",
    rotation: 2
  });
	
});
```

Creates the following output: 
```
UPLOADING: Uploading project bundle to IoT device.                                                                                                                                                                                     
[ Upload Complete ]                                                                                                                                                                                                                    
Intel XDK - Message Received: run                                                                                                                                                                                                      
MRAA Version: v0.5.4-110-g459ecc0                                                                                                                                                                                                      
1423265106725 Device(s) Galileo-IO (Intel Edison)                                                                                                                                                                                      
1423265106771 Connected Galileo-IO (Intel Edison)                                                                                                                                                                                      
1423265106774 Repl Initialized                                                                                                                                                                                                         
>> ERROR: node: /usr/include/node/node_buffer.h:76: static char* node::Buffer::Data(v8::Handle<v8::Value>): Assertion `val->IsObject()' failed.   
```

This is an intel edison hooked up to and Intel Edison + Arduino breakout board. I have 1 (HT16K33) with two 8x8 panels currently wired up through a breadboard. 
Any recommendations or suggestions as to how to get this working would be greatly appreciated.

rwaldron, 2015-02-07:
Thanks for the report. This may be a duplicate of (or at least related to) https://github.com/rwaldron/johnny-five/issues/611, but it may be its own issue. The first thing you might try is to install the versions posted here: https://github.com/rwaldron/johnny-five/issues/611#issuecomment-73327722

Another thing that's been on my mind: that issue _seems_ to be centered around code that would result in the `mraa` bindings being require()'ed more than once. I'm wondering if that specifically is causing problems. 

Let me know how the suggestion above goes.

rwaldron, 2015-02-07:
Update

- This is **not** a duplicate
- The I2C api for mraa bindings changed about a month ago
- The changes would directly cause this crash
- Patch to Galileo-IO/Edison-IO is coming

rwaldron, 2015-02-07:
Update

- Confirmed the example script with the WIP patches on a Galileo. 
- Next will confirm the example script with WIP patches on Edison.

rwaldron, 2015-02-07:
Update 

- Confirmed the example script with WIP patches on Edison
- May have resolved https://github.com/rwaldron/johnny-five/issues/611 in the process.

closed 2015-02-07 by rwaldron

rwaldron, 2015-02-07:
Pushed a new version of Galileo-IO/Edison-IO

nickalto, 2015-02-24:
Thank you so much. This fixed it.

rwaldron, 2015-02-24:
No problem!