rwaldron/johnny-five #614: Crash when running nodeschool nodebots course

bryan-m-hughes, 2015-02-07:
I ran a workshop yesterday where we used the nodeschool nodebots course. We consistently got the following crash when running ```nodebot-workshop verify robot.js```:

```
ReferenceError: isOnBoard is not defined
    at new Board (/usr/local/lib/node_modules/nodebot-workshop/node_modules/johnny-five/lib/board.js:294:9)
    at Object.<anonymous> (/Users/bryan/Projects/nodebotsday/robot.js:2:13)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Object.require.extensions.(anonymous function) (/usr/local/lib/node_modules/nodebot-workshop/node_modules/proxyquire/lib/proxyquire.js:246:43)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at Proxyquire._withoutCache (/usr/local/lib/node_modules/nodebot-workshop/node_modules/proxyquire/lib/proxyquire.js:180:12)
    at Proxyquire.load (/usr/local/lib/node_modules/nodebot-workshop/node_modules/proxyquire/lib/proxyquire.js:135:15)

```

Here's the code

```
var five = require("johnny-five"),
    board = new five.Board();
```

We noticed this with Johnny-Five 0.8.40 on OS X, Linux, and Windows.

rwaldron, 2015-02-07:
Thanks, that was a `-dev-00X` release being used to test on remote Edison devices and was fixed last night. The `nodebot-workshop` should be locked to a specific version and periodically updated.

This is really frustrating for me—anyone that was at the workshop will likely think that Johnny-Five is some broken garbage framework that's either just poorly maintained, or too hard to use, or both. I'd like to think that none of those things are true.

closed 2015-02-07 by rwaldron

bryan-m-hughes, 2015-02-07:
So we actually saw it on all 0.8.x releases. I tried as far back as 0.8.30, and tried 0.8.41 (non ```-dev-00X```) this morning and the issue persists, but only in the workshop, not on actual hardware.

rwaldron, 2015-02-08:
Respectfully, that's simply not possible. The history of all occurrences of `isOnBoard` in board.js: 

- v0.8.41
  - https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L28-L39
  - https://github.com/rwaldron/johnny-five/blob/master/lib/board.js#L294-L297

- [v0.8.41-dev-001](http://registry.npmjs.org/johnny-five/-/johnny-five-0.8.41-dev-001.tgz)
  - L28-L39
  - L294-L297 (Not commented out here - this is the source of the crash)
    - Note that this is the line that the error is found at: `at new Board (/usr/local/lib/node_modules/nodebot-workshop/node_modules/johnny-five/lib/board.js:294:9)`

- [v0.8.40](http://registry.npmjs.org/johnny-five/-/johnny-five-0.8.40.tgz)
  - L28-L39 (Not commented out here)
  - L294-L297 (Not commented out here)

...

- [v0.8.30](http://registry.npmjs.org/johnny-five/-/johnny-five-0.8.30.tgz)
  - L25-L36 (Not commented out here)
  - L288-L302 (Not commented out here)

dtex, 2015-02-08:
> Here's the code
> ```
> var five = require("johnny-five"),
>    board = new five.Board();
> ```

That's not all the code is it? What up with Proxyquire in the stack trace?

rwaldron, 2015-02-08:
That would be enough to crash it, but the issue simply cannot still exist. The only place that `isOnBoard` exists, is in a comment.

bryan-m-hughes, 2015-02-08:
No you're absolutely right...I have no idea what is going on in my system right now. I added a single line of code to print the version grabbed from johnny-five's package.json, and it all of a sudden it started working, and now I can't reproduce (to be clear, I did not reinstall johnny-five)...so...wtf node?

Whatever, it's working now and you are correct that the bug has been fixed.

rwaldron, 2015-02-08:
That's a relief—I was worried that there was some deeper issue (stranger things have happened, remind me to tell you about the Galileo bug the next time I see you).