rwaldron/johnny-five #615: Intel Edison: Sensor, using on() crashes.

mpbod, 2015-02-08:
on the Edison, trying to get Sensor working.
```js
this.titl_sensor = new five.Sensor({
    pin: "A1",
    freq: 100
});
this.titl_sensor.on("data", function() {
    console.log(this.value);
});
```
Result:
```
ERROR: undefined:0                                                                                                                                                                

ERROR: TypeError: undefined is not a function
```
this = board – inside the board's on ready function call

I can read the value directly, but no when using `on()` it crashes like the above...

rwaldron, 2015-02-09:
Arduino Board or Mini Board?

rwaldron, 2015-02-09:
Possibly related: https://github.com/intel-iot-devkit/mraa/issues/117#issuecomment-73426680 ?

rwaldron, 2015-02-09:
The program appears to run fine for me, but will crash when I hit a key on the keyboard.

mpbod, 2015-02-09:
Seems to be working now...I'm not sure what was wrong. Closing. Thanks for investigating.

closed 2015-02-09 by mpbod

rexstjohn, 2015-02-09:
@rwaldron @mpbod - Had some problems this weekend at VTHacks with the Edison image ending in "_15." When we used the image ending in "_14" things seemed to work. I suspect this may be related. Also, on the 14 image, had to update libMRAA manually to make sure it worked or analog sensors always returned 0.

Try maybe using the Edison image here: https://drive.google.com/folderview?id=0B6gHgawzKtxCM29zOHdSckh4em8&usp=sharing&tid=0B6gHgawzKtxCV2M3THJvaTFQek0

mpbod, 2015-02-09:
@rexstjohn I'm on _14 at the moment. Will try with _15 and check this sometime these few days.

rwaldron, 2015-02-09:
@rexstjohn interesting, thanks for the details