rwaldron/johnny-five #616: Issues when running motor-PCA9685.js example

ashishdatta, 2015-02-09:
Hello I am using Intel Edison. When I run the example code (pasted below) I get this output: 

```bash
1423459315679 Device(s) Galileo-IO (Intel Edison) 
1423459315723 Connected Galileo-IO (Intel Edison) 
1423459315725 Repl Initialized 
node: /usr/include/node/node_buffer.h:76: static char* node::Buffer::Data(v8::Handle<v8::Value>): Assertion `val->IsObject()' failed.
Aborted
```

The only change that I made from ```motor-PCA9685.js``` is adding 

```javascript
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

```

I made this change due to @mpbod fork documenting the necessary change. Nothing is being reported to ```journalctl -f```

# Code
```javascript
var motor, led;
var five = require("johnny-five");
var Edison = require("galileo-io");
var board = new five.Board({
  io: new Edison()
});

/*
 * The PCA9685 controller has been test on
 * the Adafruit Motor/Stepper/Servo Shield v2
 */

board.on("ready", function() {
  // Create a new `motor` hardware instance.
  motor = new five.Motor({
    pins: [8, 9, 10],
    controller: "PCA9685",
    address: 0x60
  });

  // Inject the `motor` hardware into
  // the Repl instance's context;
  // allows direct command line access
  board.repl.inject({
    motor: motor,
  });

  // Motor Event API

  // "start" events fire when the motor is started.
  motor.on("start", function(err, timestamp) {
    console.log("start", timestamp);

    // Demonstrate motor stop in 2 seconds
    board.wait(2000, function() {
      motor.stop();
    });
  });

  // "stop" events fire when the motor is started.
  motor.on("stop", function(err, timestamp) {
    console.log("stop", timestamp);
  });

  // Motor API

  // start()
  // Start the motor. `isOn` property set to |true|
  motor.start();

});

```

Details about Edison:

### MRAA Version
```bash
Version: 0.5.4.110
```
### Galileo-IO Version
```bash
v0.8.1
```
### Intel Edison Version
```bash
Version #120 (configure_edison --version) 
```

I did recently upgrade to the newest Edison image...but I also get the error on a 2nd to latest image (ww42). Any help would be appreciated, thanks!

rwaldron, 2015-02-09:
Strange—that's the same error here: https://github.com/rwaldron/johnny-five/issues/612 but this was fixed yesterday.

ashishdatta, 2015-02-09:
I uninstalled galileo-io and reinstalled and it worked, must have had an older version conflicting somewhere. Thanks.

closed 2015-02-09 by ashishdatta

rwaldron, 2015-02-09:
That's great news :) I hadn't had a chance to get to this yet, so I'm glad that it worked out.