rwaldron/johnny-five #618: TypeError: Cannot read property 'OUTPUT' of undefined

Christopheraburns, 2015-02-11:
Sorry for the newbie question here on github, Can someone point me in the right direction to resolve this?

I believe I have all the necessary prereqs in place.  

Thanks!
[image]

rwaldron, 2015-02-11:
Can you provide the following: 

- Code that caused this
- Version of Johnny-Five
- Board used

Christopheraburns, 2015-02-11:
The code is the basic board initialization example:

var five = require("johnny-five");
var board = new five.Board();

// The board's pins will not be accessible until
// the board has reported that it is ready
board.on("ready", function() {
  // Set pin 13 to OUTPUT mode
  this.pinMode(13, this.MODES.OUTPUT);

  // Create a loop to "flash/blink/strobe" an led
  this.loop(500, function() {
    // Whatever the last value was, write the opposite
    this.digitalWrite(13, this.pins[13].value ? 0 : 1);
  });
});

*johnny-five version = 0.8.42
*There are actually two boards I am testing with:
Arduino Yun
Arduino Uno
I have tried one at a time, both together.

*I am running a windows 7 os
*The boards are connected via a powered USB hub but they show up in Device manager as COM8 and COM14.
*Standard Firmata successfully uploaded to both boards.

Thanks!

rwaldron, 2015-02-11:
> Standard Firmata successfully uploaded to both boards.

That won't work on the Yun and to be honest, getting the Yun set up correctly is simply not worth the headache. A Yun running LininoIO can just run Johnny-Five directly on the board itself, no Firmata. Just SSH in and then `npm install johnny-five nino-io`, then this should work: 

```js
var five = require("johnny-five");
var Nino = require("nino-io");
var board = new five.Board({
  io: new Nino({
    layout: "Arduino Yun"
  })
});

board.on("ready", function() {
  console.log("Ready");

  var led = new five.Led(13);

  led.blink(500);
});
```

------

The issue is just that there is a typo "bug" in that example. I updated the example file and simplified it: https://github.com/rwaldron/johnny-five/blob/master/docs/board.md

Thanks for reporting!

closed 2015-02-11 by rwaldron