rwaldron/johnny-five #621: Running johnny five programm without laptop

stefanvermaas, 2015-02-13:
Hi guys,

It might be a silly and dumb question, but is it also possible to upload to program to the arduino? So it can run without interference of a computer? I can't find any documentation about this.

Cheers.

rwaldron, 2015-02-14:
> It might be a silly and dumb question

There is no such thing :P

> but is it also possible to upload to program to the arduino?

No, not the Arduino specifically, but yes you can run your Johnny-Five code directly on a BeagleBone, Raspberry PI, Linino One, Galileo or Edison (and others in progress).

https://github.com/rwaldron/johnny-five/wiki/IO-Plugins

closed 2015-02-14 by rwaldron

sant123, 2016-11-28:
Hi @rwaldron, why is not possible to do that with Arduino? I've seen Arduino working without any laptop or dependency.... Just the Arduino with something to control 🎮. It'd be really good to do that.

rwaldron, 2016-11-28:
@sant123 it's certainly possible to use an Arduino without a connection to a laptop, but it's not possible to use Johnny-Five with an Arduino, since the JavaScript must run on some capable host (which the Arduino is not)

sant123, 2016-11-28:
That sounds really sad @rwaldron, because after the program has been executed it should persist it and then it could be used without a laptop; like Arduino IDE.

By the way, thank you!

rwaldron, 2016-11-28:
The Arduino doesn't host a JavaScript runtime. If you want to execute Johnny-Five code on a device, you'll need to use a platform that runs Linux and is capable of supporting Node.js. Otherwise, it's a host/client relationship that requires some kind of tether. It's not "sad", it's just the reality of supporting a variety of disparate hardware platforms.

henricavalcante, 2016-11-28:
@stefanvermaas To run JavaScript inside a hardware I recommend to use [Tessel 2](https://tessel.io/).

stefanvermaas, 2016-11-28:
Thanks for all the info! 😬

LukasBombach, 2019-05-23:
Hey there,

I am a bit confused. How is this project intended to be used if it can only be used with a PC attached to your hardware? Is this just for drafting things? How would you go und run your Arduino based hardware if you don't want to run a Laptop next to your IOT project?

LukasBombach, 2019-05-23:
Also, what about https://www.lowjs.org/ | https://github.com/neonious/lowjs ?

dtex, 2019-05-23:
Johnny-Five is built on node.js, so it's not going to run on a humble Arduino. There are a number of single board computers that have GPIO on board and can run node.js and Johnny-Five (Raspberry Pi, Tessel 2, Beaglebone, etc) so you have a way to create a self-contained device with J5. If it has to be on a humble device like an Uno you will need a host PC connected via serial, Bluetooth, wi-fi or whatever.

lowjs looks interesting. I haven't spent any time with it so I can't comment on it.

calebdiaz, 2021-12-10:
> Hey there,
> 
> I am a bit confused. How is this project intended to be used if it can only be used with a PC attached to your hardware? Is this just for drafting things? How would you go und run your Arduino based hardware if you don't want to run a Laptop next to your IOT project?
> 
> Still I don't want to talk down on the project, I really appreciate it, I am just wondering how to use it.

Maybe you could control a robotic arm or get data from some kind of sensor interfaced trough arduino and use your computer to process data really fast with your CPU and readjust again the control.