rwaldron/johnny-five #622: Supporting L298 H-Bridge Motor Control board

dulichan, 2015-02-15:
Hi guys,
I looked around the Motor documentation and the source code to see if [L298 H-Bridge Motor controller] (http://blog.pennybuying.com/down/f/F815A.pdf) is supported. I tried to see if this was similar to the Arduino Motor Shield but seems like it has a different concept. 

There are 2 channels. Channel A has 3 pins. One pin is the PWM pin. The other 2 pins are IN1 & IN2. When IN1 is HIGH and IN2 is LOW - the motors move forward. When IN1 is LOW and IN2 is HIGH- the motors reverse. How does it map with the Johnny Five motor implementation?

Below implementation seems to be working but I am using a PIN object to set the PWM pin to HIGH. 
```
	var pinA = new five.Pin(10);
	var motor = 
		new five.Motor({
	       pins: {
	         pwm: 2,
	         dir: 3
	       }
	     });
	
	pinA.high();
	motor.forward();
```

What's the correct way to use it without using a PIN object?
Cheers~

dulichan, 2015-02-15:
Hi guys,
I figure it out. We have to use a cdir property. Below is my example.
```
var motorA = new five.Motor({
        pins: {
            pwm: channelA.pin,
            dir: channelA.dir,
            cdir: channelA.cdir
        }
    });
```

channelA.pin - ENA
channelA.dir - IN1
channelA.dir - IN2
Cheers~

dtex, 2015-02-15:
@dulichan 

Here is an example of that style motor shield.

https://github.com/rwaldron/johnny-five/blob/master/docs/motor-3-pin.md

dulichan, 2015-02-15:
Thanks @dtex

closed 2015-02-15 by dulichan

dtex, 2015-02-15:
Ha, I was two seconds too late. Glad you found it.

cookiengineer, 2017-05-09:
For the sake of clarity when others are googling for this issue, as it took me around 3 hours of shitty internet tutorials to figure out:

The wiring of the L298N for _two_ motors is different than expected or explained in all related issues.

PWMs that are used (outer pins on L298N) are `Pin 9` and `Pin 10` on the Arduino. `Pin 2 to 5` are the digital control channels (inner pins on L298N).

With my setup it's also important to know  that the `Voltage Jumper` on the L298N being connected means it runs in `7.5V - 8V` for my DC motors. Less voltage (all tutorials say 5V) is not working.

Might be obvious to advanced users, but unknown to noobs like me: If you don't have a common ground connected to the L298N AND the Arduino's GND Pin (next to pin 13), the L298N will stay dead and don't react to the PWMs as they need a common GND.

```javascript
const _five = require('johnny-five');
const _board = new _five.Board({
    port: '/dev/ttyUSB0' // change on OSX or Windows
});

_board.on('ready', function() {

    let motor_l = new _five.Motor({
        pins: {
            pwm: 9,
            dir: 2,
            cdir: 3
        }
    });

    let motor_r = new _five.Motor({
        pins: {
            pwm: 10,
            dir: 4,
            cdir: 5
        }
    });

    motor_l.forward(255);
    motor_r.reverse(255);

});
```

dtex, 2017-05-09:
Hello @cookiengineer 

It sounds like you were using the L298N IC directly and not a breakout board or shield. So, one of these thingies:
[image]
If so, it sounds an example for the L298N that is similar to [this on for the SN754410](http://johnny-five.io/examples/motor-hbridge/) would have prevented your frustration correct?