rwaldron/johnny-five #623: Wrong data if multiple sensors are used at once

ritz078, 2015-02-17:
Hi,

I am using a proximity sensor and temperature sensor connected to a arduino uno. When only one is connected they work fine and even when both are connected and proximity sensor has nothing in between, the sensors work fine but the moment there is something between the proximity sensor, there are abrupt changes in the temperature sensor. below is my code ;

  ```javascript
board.on("ready", function() {
    var temperature = new five.Temperature({
      controller: "LM35",
      pin: "A4"
    });

    temperature.on("change", function(err, data) {
      io.emit('temperature',data.celsius);
      console.log(data.celsius+' C')
    });

    var proximity = new five.IR.Proximity({
      controller: controller,
      pin: "A0"
    });

    proximity.on("change", function() {
      io.emit('proximity',this.inches);
      console.log("inches: ", this.inches);
    });
  });
```

rwaldron, 2015-02-17:
```js
    var proximity = new five.IR.Proximity({
      controller: controller,
      pin: "A0"
    });
```

There is no controller specified :)

What model is your IR proximity sensor? For example, if it's `GP2Y0A21YK`, then you'll want to initialize it like this: 

```js
    var proximity = new five.IR.Proximity({
      controller: "GP2Y0A21YK",
      pin: "A0"
    });
```

ritz078, 2015-02-17:
it is another sensor model but separately its working fine.

Resseguie, 2015-02-17:
Note that there was discussion about this on gitter for anyone that wants to see background chatter:
https://gitter.im/rwaldron/johnny-five?at=54e31b1051b516c761d7999a (and following)

Thought from @ajfisher is that it's an ADC issue and having time to stabilize caps between reads? (He can chime in with more details)

rwaldron, 2015-02-17:
I can't reproduce this. 

Using: 
- LM35
- GP2Y0A21YK

(I have every available Sharp IR here, so please tell me which)

ajfisher, 2015-02-17:
Make sure you're using two entirely analog sensors rather than an I2C one. I'm not sure which was @ritz078 was using but I don't believe it was the Sharp one.

ajfisher, 2015-02-18:
As follow up - I don't have access to either of these sensors where I am, however using the Sensor object x2 with simple potentiometers in order to replicate two analog inputs being read at high speed, I can't replicate this.

I would suggest @ritz078 tries a test case of using the Sensor object for both components rather than Temperature and Proximity in order to ensure that the data is coming through and is stable. Something like this should work:

```
 var five = require("johnny-five");                                              

var board=new five.Board();                                                     
board.on("ready", function() {                                                  
   var temperature = new five.Sensor({                                         
     pin: "A0",                                                                
     freq: 50,                                                                 
   });                                                                         

   temperature.on("data", function(err, data) {                                
     console.log("A0: " + this.value)                                          
   });                                                                         

   var proximity = new five.Sensor({                                           
     pin: "A1",                                                                
     freq: 50,                                                                                                                                    
   });                                                                         

   proximity.on("data", function() {                                           
     console.log("A1: ", this.value);                                          
   });                                                                         
 }); 
```

This would most closely replicate the arduino sketch we played around with that was found to be working.

ritz078, 2015-02-18:
Its a H505 photoelectric sensor. Now i am using MPU 6050 instead of LM35 but still there are abrupt changes  in the MPU readings while the H505 readings are always correct when both are used simultaneously and the abrupt change only happens when the readings of H505 are changed.
Do you think its an issue with H505 ?

rwaldron, 2015-02-18:
> Its a H505 photoelectric sensor.
> Do you think its an issue with H505 ?

Seems likely. Where can I order this from? Are any of these what I'm looking for: 

- http://www.amazon.com/Photo-Interrupter-Slotted-Optical-Switch/dp/B008DG97KW
- http://www.alibaba.com/product-detail/Photoelectric-Switch-H505_231158106.html

Just to be clear, Johnny-Five doesn't know what that is (yet). Here are the supported analog IR proximity sensors: https://github.com/rwaldron/johnny-five/wiki/IR.Proximity . In the meantime, using the `Sensor` class will be fine (while we figure out what's wrong).

ritz078, 2015-02-19:
Yes  those links for H505 are correct. I tried with the sensor class but the problem was still there.

ajfisher, 2015-02-19:
If that's the case then maybe the H505 requires some sort of stabilisation
before the read?

ritz078, 2015-02-19:
No, I don't think so because in that case there would have been fluctuations in H505's data and not LM35 but here H505's data is totally correct.

rwaldron, 2015-02-19:
@ritz078 ok, thanks. The next step is that I need to acquire one of these if I'm to be any help to you. I appreciate your patience so far :) I've just ordered the first one on Amazon and will receive it tomorrow. Once it arrives, I will be able to help you more effectively :D

ritz078, 2015-02-19:
Thanks @ajfisher and @rwaldron for all the help :)

rwaldron, 2015-02-21:
@ritz078 I received the sensors, can you post a picture of your wiring?

ritz078, 2015-02-22:
[image]
Here's the image :)

rwaldron, 2015-02-22:
Ok, this is very helpful, but I just learned that I don't have the right component. I got _just_ the sensor :\ Can you link me to the exact component shown in this picture? Thanks!! (and also: thanks for your patience!!)

ritz078, 2015-02-22:
I have one extra . If you want I can send it to you.

rwaldron, 2015-02-22:
No, that won't be necessary—just send me the link :)

Rupaliitr, 2015-02-24:
It's a custom made sensor so I can't find the link.You can use the same sensor H505 which you have. In that sensor you only have diode and phototransistor.
Here is the circuit diagram.
[image]

Connect a 331 resistor chip (330 ohm) in series with diode pins ( on the top of sensor you can see diode symbol) and 103 resistor chip in series with phototransistor (same as given in circuit diagram). links are here
http://www.aliexpress.com/store/product/1206-330E-5-Surface-printing-331-Chip-Resistor/903701_585435200.html

https://www.dipmicro.com/store/R4Y100

I hope this will work :)

rwaldron, 2015-02-24:
That's extremely helpful, thanks!!

rwaldron, 2015-02-25:
Using the above circuit, I made a simulation with iCircuit: 

[image]

Then I assembled it: 

[image]

Along with an Led and a Thermistor: 

[image]

And used this code: 

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var led = new five.Led(8);
  var temp = new five.Temperature({
    controller: "GROVE",
    pin: "A2"
  });
  var sensor = new five.Sensor("A0");

  sensor.on("change", function() {
    if (this.value < 25) {
      led.on();
      console.log("Slot interrupted @ %d", Date.now());
      console.log("Temperature: %d°C", Math.round(temp.celsius));
    } else {
      led.off();
    }
  });
});
```

Here's a video demo: https://www.youtube.com/watch?v=7BHT7QUWMuY

And here's a sample of console output: 

```
$ node eg/hy505.js
1424902063749 Device(s) /dev/cu.usbmodem1421
1424902063761 Connected /dev/cu.usbmodem1421
1424902067993 Repl Initialized
>> Slot interrupted @ 1424902069419
Temperature: 24°C
Slot interrupted @ 1424902069873
Temperature: 24°C
Slot interrupted @ 1424902074362
Temperature: 24°C
Slot interrupted @ 1424902075577
Temperature: 24°C
Slot interrupted @ 1424902075605
...
```

Based on this, I think you might need to take another look at the HY505 hardware in your circuit—keep us posted!

wardialer, 2015-08-04:
If i use my LM35 Temperature with this code, the sensor works fine: 

    var five = require("johnny-five");
    var board = new five.Board();
    board.on("ready", function() {
      var temp = new five.Temperature({
        pin: "A0",
        controller: "LM35"
      });

    //  var sensor = new five.Sensor("A5");

      temp.on("change", function() {
        console.log("Temp: %d", this.celsius);
      });
    });

But, when i simply uncomment row 9: 

    var sensor = new five.Sensor("A5");

my temperature sensor starts printing wrong values, changing from ~24°C to ~50°C in seconds. 
The circuit remains the same. In the A5 pin is attached a terrain humidity sensor, both sensors works fine alone (i.e. attached to my arduino but undeclared on code) but when i put them together temperature sensor goes crazy.
How can this happen?

rwaldron, 2015-08-04:
@wardialer I need more information, because I can't reproduce this issue—however I'm using a TMP36, not an LM35. I just ordered an LM35.

- What board are you using?
- What, if anything, is attached to A5?

ajfisher, 2015-08-04:
It might be because the voltage is wavering on the LM35 if you're reading
it directly. Make sure you've got the 2k series resistor in place in your
circuit.

wardialer, 2015-08-05:
@rwaldron I'm using an Arduino Uno. If nothing is attached to A5, all works fine.
@ajfisher I'll try later (i'm at work right now), but it seems strange to me, because the circuit works fine if i don't initialize the sensor in my code... Thats why i thought it was a software problem

rwaldron, 2015-08-05:
>  If nothing is attached to A5, all works fine.

With or without the `var sensor = new five.Sensor("A5");` line?

wardialer, 2015-08-05:
With and without the line

rwaldron, 2015-08-05:
So what thing is connected to A5 when the values erratically change?

wardialer, 2015-08-05:
It was a homemade terrain-humidity sensor like this: 
[image]

and a photoresistor. Here you have the complete circuit (here the sensor is attached to A4 as a test...)
[image]

rwaldron, 2015-08-05:
is that what the yellow and red wires lead to?

wardialer, 2015-08-05:
yes, sorry for bad picture

rwaldron, 2015-08-06:
The picture is just fine, thank you for taking the time to produce it for the big report. I just wanted to make sure I understand the relationship between the two images :)

Now I have another question, that I should've asked before and I apologize for not doing so: which wire in the photo goes to which connection in the diagram?

Orange -> 5V ?
Red -> A0?

wardialer, 2015-08-06:
Red wires are connected to 5v for both images.The blue cable in the schema is the orange one in the picture and is attached to A5 instead of A0.

reconbot, 2016-08-24:
I'm going to close this issue due to it's age, but if you'd like to continue with it feel free to comment and we'll reopen. I hope you got it working!

closed 2016-08-24 by reconbot

FranciscoHanna, 2016-09-30:
Hi. I would like this thread to be reopen since I couldn't find a solution to a similar problem I'm experimenting.

Right now I have two photoresitor modules attached to A0 and A1. Whenever the value in one photoresitor changes, it affects the other. From reading, I've come to the conclusion that it is a ADC problem, since the ADC needs time between readings so that a current reading doesn't affect further readings in analog inputs. In a pure arduino sketch, this is solved by placing a delay between readings or by reading the same input twice with delay between this readings. So, a way to solve this in J5 would be adding this delay. I couldn't acomplished this using 'freq' option in Sensor Class (I don't thinks this option it's meant to acomplished this). Any ideas on this?

Here is my code
`var five = require("johnny-five");  

var board = new five.Board();  

board.on("ready", function() {  
    console.log('Arduino connected');

    var sensorA = new five.Sensor({pin: "A0", freq: 157});
    var sensorB = new five.Sensor({pin: "A1", freq: 313});

    sensorA.on("data", function(){
        console.log("A val:" + this.value);
    });

    sensorB.on("data", function(){
        console.log("B val:" + this.value);
    });
});`

rwaldron, 2016-10-04:
> From reading, I've come to the conclusion that it is a ADC problem, since the ADC needs time between readings so that a current reading doesn't affect further readings in analog inputs

@soundanalogous any thoughts on this?

soundanalogous, 2016-10-05:
Strange how this issue has never come up in all the years Firmata has been around. Anyway, a delay after [this line](https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L800) would fix it. However, deciding the duration of that delay is not so simple because Firmata can run on several different microcontroller architectures and this delay likely differs to some extent per architecture. I wouldn't want to set the delay to be the longest required delay for a particular architecture to be applied to all architectures.

soundanalogous, 2016-10-05:
Interesting... the Arduino analogRead function actually [blocks until the conversion is finished](https://github.com/arduino/Arduino/blob/62b723357c2866935e108945f5846573df8c1be1/hardware/arduino/avr/cores/arduino/wiring_analog.c#L80) so a delay should not be necessary.

@FranciscoHanna, I have a few questions for you:

1. What board are you using?
2. How do you have the photoresistors wired to the board?
3. Can you try adding a short delay just after this line: https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L800 and see if that makes a difference.

FranciscoHanna, 2016-10-05:
@soundanalogous Let me answer those questions:

1. I'm using an Arduino Uno. It's probably a chinesse replica because here in Argentina is not easy to find the originals.
2. I'm using [this](http://img.dxcdn.com/productimages/sku_151310_2.jpg) photoresistor module so the wiring is pretty staright forward.
3. I'll try this on weekend and I'll come back and post the results.

webondevices, 2016-11-28:
I have a similar issue when using an LDR and a LM35 circuit simultaneously. One connected to A0 the other to A1.

They work perfectly individually but as soon as I try using both of them my temperature readings starts fluctuating from 15 to 50 celsius when it should be 20 degrees.

webondevices, 2016-11-28:
So I tried modifying the StandardFirmata sketch by performing a dummy analogRead() before the actual Firmata read that is being sent through serial and it has solved my problem!
```
dummyRead = analogRead(analogPin);
Firmata.sendAnalog(analogPin, analogRead(analogPin));
```