rwaldron/johnny-five #624: SPI support

natevw, 2015-02-18:
Would be cool if johnny-five had SPI support, it would enable compatibility with libraries like https://github.com/natevw/node-nrf and https://github.com/tessel/sdcard.

I think we've talked about this off and on through the years, and I know Firmata lacks [or lacked?] support for the SPI protocol, but I couldn't find reference to that on this project. So I'm mostly filing this as a placeholder I can cross-reference, and of course it'd be great if this becomes a rally point for an implementation too!

bryan-m-hughes, 2015-02-18:
We were discussing this very thing in [gitter](https://gitter.im/rwaldron/johnny-five) the other day IIRC (can't seem to find it at the moment). I think the idea was that someone would throw together a proposal and implement it in one of the non-firmata based platforms first to see how things look and go from there.

soundanalogous, 2015-02-18:
I'm beginning work on SPI for Firmata: https://github.com/firmata/protocol/issues/26.

I also have a Galileo rev 2 on order so I can work on defining a SPI interface for J5 that will work both on Linux and Arduino.

rwaldron, 2015-02-18:
Moving to: https://github.com/rwaldron/io-plugins/issues/3

closed 2015-02-18 by rwaldron