rwaldron/johnny-five #625: Design integration testing guide

rwaldron, 2015-02-18:
From an offline conversation with @daniellmb: 

- Show developers a clear roadmap for how to unit test the applications they write with the framework. 
- Examples of how to do TDR/TDD 
- End-to-end integration testing. For example: teach your bot to type then run a spec that goes through the alphabet and verifies that when the letter "a" should be typed, X milliseconds that input is received from the keyboard.

@divanvisagie tell me your thoughts :)

divanvisagie, 2015-02-22:
@rwaldron TDR being Test Driven Robotics? I like the sound of that. Don't do it much outside of contributions though. Anyway I'm back in the land of the living after a crazy after hours project I hurled myself into so I should be able to help out again.

This sounds like it can be split onto two sections in the wiki , TDR and End-to-end

Maybe like
- Test Driven Robotics (with examples) 
- Integration Testing (with examples)

And we tackle one at a time?

McDonnellDean, 2015-02-22:
I've been in hiding due to moving careers from .net to node but it should all settle down soon enough and the crowd I'm moving to (Nearform) seem to have at least an interest in IoT so hope to be a bit more active.

In terms of TDR that sounds really awesome I'll be keeping a close eye and will jump in were possible.

rwaldron, 2015-02-22:
@divanvisagie sounds good and you should lead the effort :)

mMerlin, 2015-05-07:
Is/has anything happening/ed with this?  I searched the wiki but did not find anything about TTD, TDR, testing.

I have been looking for information about getting started with johnny-five projects and nodeunit.  I started by using "grunt-init node" to get the initial structure, then copied mock-firmata.js and mock-pins.js from the test/util folder to my own project, plus added board-io to the dev dependences in package.json.  That has been enough for simple test cases.  I saw a post about making MockFirmata its own project / repository, but that does not seem to have been done yet.

I expect more will be needed to test with event handlers.

rwaldron, 2015-05-07:
Nothing has happened yet, but not because no one cares ;)

I'm not really sure how to solve this, or even how to approach it. My first and most naive thought would be to produce Mock-Johnny-Five and publish it as a module. It would have all the stuff Johnny-Five has, but everything would be empty and ready for use as a mock. "Mock-Johnny-Five" would be generated and maintained by a build process, such that any time Johnny-Five is published, so is Mock-Johnny-Five. Maybe that's not enough? I'm not sure.

closed 2018-08-28 by stale[bot]