rwaldron/johnny-five #627: Jhonny-Five < -- > Ethernet < -- > Arduino

henricavalcante, 2015-02-19:
Can I use something like this to connect to Arduino?

https://github.com/peterschwarz/standard-firmata-ethernet

soundanalogous, 2015-02-20:
That library uses an old version of StandardFirmata so I wouldn't recommend it, but it will work with the version of Firmata you already have installed in your Arduino application. You can use [ConfigurableFirmata](https://github.com/firmata/arduino/releases/tag/v2.6.2) to use Firmata over Ethernet but it requires installing a new version of Firmata (which is recommended).

However, [node-firmata](https://github.com/jgautier/firmata), which is what johnny-five uses to interface with an Arduino board running Firmata, currently only supports a Serial interface, you'd have to update it to support Ethernet.

To use ConfigurableFirmata with Ethernet, [see this gist](https://gist.github.com/soundanalogous/3d6d3d3f76e3a529e2cc) and follow the instructions posted in the first comment.

henricavalcante, 2015-02-20:
Ok @soundanalogous, thanks for the light, I'll try to do something.

rwaldron, 2015-02-22:
This thread has inspired me to finally work on making Firmata.js capable of using alternative communication transports. Hopefully will have something by tomorrow :)

monteslu, 2015-02-22:
Only issue I've had using different communication transports with firmata is that johnny-five doesn't currently emit a ready event without flagging the firmata instance with isReady=true
This works now: https://gist.github.com/monteslu/6ef34e8ec01e7e5675e2

Pretty much any node stream or [virtual serial port](http://blog.iceddev.com/2014-12-05-remotebots-cutting-the-cord-with-remote-io.html) will work.

soundanalogous, 2015-02-22:
I can work on StandardFirmataEthernet.

rwaldron, 2015-02-22:
@monteslu tried your gist with ConfigurableFirmata (with remote_ip, local_ip set) and it won't connect, it's possible I'm doing something wrong, but I didn't spend much time with it. 

I've been working on `EtherPort`, which is a wrapper around some logic that creates a socket with  `net.createServer`. Connecting executes perfectly, but nothing happens when I try to digital write (led blinking of course)

rwaldron, 2015-02-22:
https://github.com/rwaldron/etherport

monteslu, 2015-02-22:
@rwaldron sorry about that didn't mean to imply there isn't work to do on the arduino side. Just that not a lot had to change on the firmata.js side for it to work.

 Still need to channel the ethernet data to what would otherwise be serial input on the device. @jacobrosenthal has done a bunch work in this area on various sketches.

soundanalogous, 2015-02-23:
I've created a version of StandardFirmata with Ethernet that is working with Rick's Etherport example.

Assumes you have an Arduino Uno or Arduino Ethernet board or Ethernet shield (or other compatible ethernet board / shield).

1. Checkout the [ethernet branch](https://github.com/firmata/arduino/tree/ethernet) from firmata/arduino, run release.sh and copy the contents of the appropriate .zip file to your Arduino core libraries folder, overwriting the existing Firmata library.
2. Open the StandardFirmataEthernet (SFE) example and perform the following:
  - Change the IP address on line 39 to match the IP address of the computer running the etherport example
  - Either comment out line 48 or set the IP address if your Ethernet  shield (both options are working for me)
  - Update line 50 to add the mac address of your Ethernet shield
3. Compile and upload
4. Open eg/blink.js in the etherport repo and change the pin number to something other than 10, 11, 12 or 13 (I used 2... submitted a pull request for same change) and wire an LED to that pin.
5. Run eg/blink.js in the etherport repo

I'm using Norbert Truchsess' EthernetClientStream which will maintain a connection and reconnect if dropped. This also enables the arduino client to be running before the node server is started and vice versa.

rwaldron, 2015-02-23:
@soundanalogous successful runs with the "blink.js" example in EtherPort, with digital pin 2: 

- StandardFirmataEthernet.ino-2.4 
- ConfigurableFirmata.ino-2.6

mcinnes01, 2015-03-09:
@rwaldron thanks for your clear steps I've been looking to be able to do this for ages and within about 20 minutes I'm able to control my arduino over ethernet with johnny-five. I've been trying to do this with octoblu and microblu_mqtt for ages but to no avail. Are there likely to be some features coming soon and in terms of stability where would you say this is up to? Like octublu I like the idea of being able to assign UUID's and tokens makes the connecting and addressing process a little easier/more secure. Resolving to name servers would also be really cool.

Many thanks

Andy

rwaldron, 2015-03-09:
@mcinnes01 are you using [etherport](https://github.com/rwaldron/etherport)?

mcinnes01, 2015-03-09:
@rwaldron I am indeed, its great well done!

rwaldron, 2015-03-09:
Excellent :)

closed 2015-03-09 by rwaldron

DonPancoe, 2015-06-04:
I can't get the blink.js test to work with johnny-five and etherport. When I try to run it, I get a connection timeout as shown below. I have StandardFirmataEthernet 2.4.3 flashed on the board (an Atmega328P-based [DFRobot XBoard V2](http://www.dfrobot.com/index.php?route=product/product&product_id=564#.VW-3W89VhBc)). I can ping the board's IP address, but when I try PortQry at that address on port 3030, I get "FILTERED" ("LISTENING" is what I'd hope to see).

I am using Arduino IDE 1.6.4, and have node.js, johnny-five and etherport installed. Any help would be appreciated as I'm diving into a number of new topics at once, here.

**Example command line output:**
C:\Program Files\nodejs\nodebot>node blink.js
1433382815040 SerialPort Listening on port: 3030
1433382815043 Connected Listening on port: 3030
1433382825045 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (C:\Program Files\nodejs\node_modules\johnny-five\lib\board.js:406:26)
    at Timer.listOnTimeout (timers.js:110:15)

C:\Program Files\nodejs\nodebot>

soundanalogous, 2015-06-04:
As long as the Wiznet chip on the XBoard is wired exactly the same way to the ATMega328 as it is with an Arduino Ethernet shield, then the XBoard should work. If SS is wired to a different pin on the XBoard than it is with an Arduino ethernet shield, then you may have an issue.

Did you make the appropriate changes in [lines 101 - 119 of StandardFirmataEthernet](https://github.com/soundanalogous/Breakout/tree/master/custom_examples)?

soundanalogous, 2015-06-04:
Actually the SS pin shouldn't make a difference here since there is only one SPI device in use.

DonPancoe, 2015-06-04:
Thanks for the reply.

I did make the necessary changes in StandardFirmataEthernet, but you might be onto something with your pin number question. The DFRobot board does match the standard Arduino pinouts, but the TQFP32 version of the Atmega328P does have a different pin numbering scheme than the DIP28 version as shown below. Could I need to modify board.h to account for this? Has anyone already done a board.h for a TQFP-based Arduino such as the Uno SMD Edition?

[image]

DonPancoe, 2015-06-04:
Also, I was curious why some of the comments (I forget which file, exactly) stated that StandardFirmataEthernet ignores requests related to D4. Does XCK/TO also have something to do with communication to the Wiznet chip? D4 is on pin 6 of the DIP28 but on pin 2 of the TQFP32. Thanks again.

soundanalogous, 2015-06-04:
There are some other Arduino compatible boards that use the TQFP32 package of the ATMega328p. I think the underlying avr-gcc library figures out that difference somehow.

The arduino ethernet shield has an SD card attached to D4 which is why it has to be ignored, but only for the ethernet shield. You can comment out lines 814, 826 and 827 in StandardFirmataEthernet.

One thing to try is running a few of the Ethernet library examples (not with J5, just on their own) in the Arduino IDE. Open File -> Examples -> Ethernet. Do they all work as expected?

soundanalogous, 2015-06-04:
Another thing to try is using the DFRobot board with johnny-five using the serial connection. Does that work without issue?

DonPancoe, 2015-06-04:
Thanks for the new info. It may take me a few days, but I will get around to trying your suggestions and reporting what I find.

rwaldron, 2015-06-04:
Sorry to jump in so late, but can you provide the exact code you're running that produces the error above? That will also be super helpful for @soundanalogous and I (or anyone else) to help you out :) Thanks!

DonPancoe, 2015-06-04:
Sorry for leaving that out. It was the test code from the EtherPort github README.md but with the LED pin changed.

var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({ 
  port: new EtherPort(3030) 
});

board.on("ready", function() {
  var led = new five.Led(13);
  led.blink(500);
});

rwaldron, 2015-06-04:
This needs to be documented, but @soundanalogous mentioned above: 

> Open eg/blink.js in the etherport repo and change the pin number to something other than 10, 11, 12 or 13 (I used 2... submitted a pull request for same change) and wire an LED to that pin.

I don't recall why this is the case, and I'm not sure if/how it matters to your case, but can you try a pin that isn't one of those?

soundanalogous, 2015-06-04:
Pin 13 is one of the SPI pins, so trying to blink that pin will mess up the communication with the Ethernet driver.

soundanalogous, 2015-06-04:
However Firmata ignores requests to change Pin 13 when using the Ethernet library so that would not cause an error here. You just wouldn't see the LED blink.

soundanalogous, 2015-06-04:
Actually it looks like Firmata is not ignoring digital writes, only set pin mode request. Try changing the LED pin to 2 or 8. If that works then I have a bug in Firmata to fix.

DonPancoe, 2015-06-05:
Quick report since it is late...

Changing the LED to pin 8 did not fix the Ethernet version. Got the same error messages as above.

The following Arduino IDE Ethernet examples all worked on the XBoard over Ethernet: UDPSendReceiveString, WebServer & ChatServer.

The following javascript blink file did work over serial with StandardFirmata flashed on the XBoard and the LED on pin 13.

var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  // Create an Led on pin 13
  var led = new five.Led(13);
  // Blink every half second
  led.blink(500); 
});

DonPancoe, 2015-06-05:
It may also be worth noting that the built-in LED on the board (pin 13) does blink two times then four times immediately after flashing the board with StandardFirmata, correctly indicating that I am using version 2.4. When I flash the board with StandardFirmataEthernet, this does not happen. Instead, the LED just glows dimly but continuously, occasionally turning off for a while then coming back on.

soundanalogous, 2015-06-05:
I just tested the same code with StandardFirmataEthernet v2.4.3 and an Arduino ethernet shield connected to an Arduino Uno. I found that I had to comment out [line 29 in index.js of the etherport module](https://github.com/voodootikigod/node-serialport#to-install) because it was throwing a TypeError on the `emit` method. After that everything was working as expected. Also had to use node v0.10 instead of node v0.12. Make sure you're using node v0.10 and that you aren't getting errors regarding the `emit` method. Otherwise the only difference between my setup and yours could possible be a difference in the bootloader that would cause a longer startup time for the DFRobot board, but the documentation says it's using an Uno bootloader. Also the fact that the Ethernet examples worked should rule out any difference in wiring.

soundanalogous, 2015-06-05:
Also, pin 13 doesn't do the version blink when using StandardFirmataEthernet. I suspect the Ethernet library sets pin 13 to digital input once the library is included so you shouldn't see it blink at all. The fact that you see the LED fading on and off makes me wonder if you are supplying enough current to your board. Try a higher current power supply if you have one, or if you're just using USB, then try a 5V 1A input instead.

rwaldron, 2015-06-05:
> Also had to use node v0.10 instead of node v0.12. 

I wonder why?

I've just ordered a [DFRobot XBoard V2](http://www.dfrobot.com/index.php?route=product/product&product_id=564#.VW-3W89VhBc)

DonPancoe, 2015-06-07:
Finally got a chance to revisit this. I've downgraded to Node.js 0.10.38, I'm powering the board with a 2A USB charger and I've commented out line 29 of index.js (*this.emit("open");*). Unfortunately, I'm still getting the command line response shown below. Is there another debug method that I'm missing? I'm not seeing anything coming over the UART.

Also, when I did downgrade to 0.10.38 and tried to npm install etherport, I got the following:
*npm WARN package.json etherport@0.1.2 No repository field.*

Thanks again for the help.

**Command Line Output**
C:\Program Files\nodejs>node blink.js
1433646895438 SerialPort Listening on port: 3030
1433646895440 Connected Listening on port: 3030
1433646905443 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (C:\Program Files\nodejs\node_modules\johnny-five\lib\board.js:406:26)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)

C:\Program Files\nodejs>

DonPancoe, 2015-06-07:
Realizing that I still had my laptop's WiFi switched on, which sometimes causes gateway conflicts with devices connected to the RJ45 Ethernet port, I tried it again with WiFi switched off. It still didn't work, but I realized that - because of how this comment system handles markups - a bit of the information being returned wasn't making it into my posts. I think I've fixed that now.

**Command Line Output**
C:\Program Files\nodejs>node blink.js
1433648154025 SerialPort Listening on port: 3030
1433648154027 Connected Listening on port: 3030
1433648164030 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

Error: A timeout occurred while connecting to the Board.
    at Board.\<anonymous\> (C:\Program Files\nodejs\node_modules\johnny-five\lib\board.js:406:26)
    at Timer.listOnTimeout \[as ontimeout\] \(timers.js:112:15\)

C:\Program Files\nodejs>

soundanalogous, 2015-06-07:
Have you tried with and without DHCP? Comment out line 115 of StandardFirmataEthernet to use DHCP. Is line 103 set to the IP address of the computer your board is connected to?

DonPancoe, 2015-06-07:
I have the laptop connected directly to the board with a single Ethernet cable, so no DHCP server is involved. I have the IP addresses of both the laptop and the board defined in the Firmata firmware. I can ping between the laptop and the board just fine, so I think the WIZnet chip is doing its job. The issue seems to be making the UDP connection to the Atmel chip and its firmware.

I will likely have to shift to a different project early next week so I may have to step away from this one for a little while. It will be interesting to see what rwaldron learns when he receives his XBoard. I've also shared this topic [over at their forum](http://www.dfrobot.com/forum/viewtopic.php?f=8&t=1432).

It sounds repetitious saying it, but I really do appreciate all your help with this issue.

rwaldron, 2015-06-09:
Quick update: I'm still awaiting my  DFRobot XBoard V2

mcinnes01, 2015-07-20:
Hey @rwaldron 

Just wondered if you've had any luck with your XBoard?

Many thanks

Andy

rodrigoadachi, 2015-07-20:
Can you use a ESP8266 as bridge, but I think a waste, since the ESP8266 is more powerful than the Arduino and even comes embedded with WiFi

DonPancoe, 2015-07-21:
Or maybe create a version of Firmata firmware to run right on the ESP8266 via the Arduino IDE. My available tinkering time is still taken up by the project I mentioned earlier, but I still hope to work on the XBoard, and maybe the ESP8266 idea, when I get a chance. Probably after the summer.

Resseguie, 2015-07-21:
@rodrigoadachi @DonPancoe the most information I'm aware of re: ESP8266 is here:
https://gist.github.com/ajfisher/5fe60fe7d8c49b3223f0

rodrigoadachi, 2015-07-21:
Resseguie, thus need a slave Arduino, the esp8266 is very powerful and low cost, if there was a Firmata for esp8266 would be perfect

rwaldron, 2015-07-21:
cc @monteslu @ajfisher

Resseguie, 2015-07-22:
@rodrigoadachi right. Believe @ajfisher was investigating just that. The above gist is his intermediate solution I believe. You might jump on gitter and check in for the details on the latest thinking.

ajfisher, 2015-07-22:
@rodrigoadachi So current status is that we are still looking to build a firmata like port across to the ESP8266. This is slightly challenging due to the nature of the timers the chip uses as well as the SDK being a currently moving target (4 major breaking releases in 6 months). 

As such the current mostly working solution is to use the ESP8266 as a wifi-serial bridge in transparent mode. This is a bit of a waste of resources given the ESP8266 is pretty powerful but does work. The set up in my gist (https://gist.github.com/ajfisher/5fe60fe7d8c49b3223f0) is known to work but it's over TCP. This is reliable and means a very straightforward implementation for johnny-five, however it is also quite problematic under reasonable message load because of the FIFO implementation on the bridge. This is a known problem.

Resulting from that there are a group of us working on a UDP based WiFi-Serial bridge. UDP is more tolerant of messages being dropped and doesn't have the ack overhead all the time which is what slows down the message processor on the TCP bridge. @monteslu has written the NodeJS side and we're just nutting out the bits on the ESP8266 side with help from the MakeBlock team (who are looking to use this in their mBots too). I'd like to say we'll have sorted this imminently as we really want to use this for NodeBots day this weekend.

As soon as we have a working solution I'll post here as well as wrap up a full gist on it explaining how to make it work

That should at least make a good wireless solution that buys time for a more dedicated implementation on the ESP8266. Though having done a lot of development on them over the last 6 months, they aren't the panacea they appear (mostly because of their lack of high resolution timers).

rodrigoadachi, 2015-07-22:
I appreciate the explanation, I know it's not a good practice, but communicate the NodeJS and esp8266 by native tcp, when opening the connection not closing, with two-way communication within 100ms packaging in the JSON at the end of each object put a ' ! ' so I can parser (.trim ('')) just so I decided my FIFO problem. Sorry if I talked nonsense or used a bad practice schedule.

rodrigoadachi, 2015-07-22:
Just a correction:

var dataString = data.toString().split('|');
          for (i in dataString){
            if (dataString[i] != ""){
              var dadaJson = JSON.parse(dataString[i]);
              // -----  
            }
          }