rwaldron/johnny-five #629: RGB LED: isAnode: true not working

nodebotanist, 2015-02-22:
When I declare an RGB LED in the following way: 

```
	var led1 = new j5.Led.RGB({
		isAnode: true,
		pins: {
			red: 2,
			green: 3,
			blue: 4
		}
	});
```

isAnode is not sticking. Debugging shows that there is no `isAnode` property on `state` when I call `write()`. 

Currently investigating a fix, will report back when I figure out what's causing it.

nodebotanist, 2015-02-22:
After messing around with the tests and the code, `state.isAnode` is coming up `undefined` in both checks for it in led.js. But the tests are passing. Working on getting the tests to reflect that `isAnode` is not being seen (i.e. getting them to fail). Then I'll work on the fix.

nodebotanist, 2015-02-22:
Found it:

```
    this.board.io.analogWrite = function(pin, value) {
      value = 255 - value;
      this.io.analogWrite(pin, value);
    }.bind(this);
```

the value = 255 - value makes the tests pass no matter what. Taking that out makes the tests fail, which is what they should be doing, I'm pretty sure. 

Working on a fix.

rwaldron, 2015-02-22:
> ```js
this.board.io.analogWrite = function(pin, value) {
  value = 255 - value;
  this.io.analogWrite(pin, value);
}.bind(this);
```

That's not in the current version of the `Led` class

rwaldron, 2015-02-22:
To pass on `isAnode`, it needs to be added here: https://github.com/rwaldron/johnny-five/blob/master/lib/led.js#L598-L603

frxnz, 2015-02-22:
> That's not in the current version of the Led class

I think that's from `test/led.js` :)

There's also a bug with the [PWM driver here](https://github.com/rwaldron/johnny-five/blob/master/lib/led.js#L66). Should be:
```js
var value = state.isAnode ? 255 - Board.constrain(state.value, 0, 255) : state.value;
```

I just ordered some common-anode LEDs to do some more thorough testing with the PWM driver. I shouldn't have relied on the tests :embarrassed:

rwaldron, 2015-02-22:
> > That's not in the current version of the Led class
> I think that's from `test/led.js` :)

I see. 

> I shouldn't have relied on the tests :embarrassed:

The tests should've been better.

closed 2015-03-02 by rwaldron

Resseguie, 2015-03-17:
I have concerns about the fix for this. It exposes `isAnode` in the underlying Led class (and generated docs). But does that even make sense? Can there be such a thing as "common anode" for a single LED? See the added example here that doesn't make sense to me: https://github.com/rwaldron/johnny-five/blob/master/docs/led-anode.md

nodebotanist, 2015-03-17:
I agree. I can take a look at this tonight when I am back at the lab.

PR #640 (2015-03-02): Add isAnode to LED state. Fixes gh-629
* [x] Add `isAnode` to Led and Led.RGB state
* [x] Fix `analogWrite` value bug in PCA9685 controller
* [x] Add test for `isAnode`
* [x] Update docs (Led.RGB has no method `pulse`)

Fixes #629 

Confirmed working with `PCA9685` and `DEFAULT` controllers