rwaldron/johnny-five #631: Proximity: consolidate all the proximity/ranging devices

rwaldron, 2015-02-23:
cc @frxnz 

- IR.Proximity: infrared proximity/ranging things.
- Sonar: Analog/I2C sonar proximity/ranging things
- Ping: Ultrasonic/pulse-based proximity/ranging things

They all produce a “distance to obstruction", which we have chosen to normalize as centimeters and provide an inch conversion.

```js
new five.Proximity({
  controller: "GP2Y0A21YK"
});

new five.Proximity({
  controller: "GP2D120XJ00F"
});

// This is the new I2C based version of course :)
new five.Proximity({
  controller: "PING" 
});

// MaxBotix MB1003 in analog mode
new five.Proximity({
  controller: "MB1003",
  pin: "A0"
});

// MaxBotix MB1202 (I2C)
new five.Proximity({
  controller: "MB1202"
});

// SRF10
new five.Proximity({
  controller: "SRF10"
});
```

Deprecate Ping, Sonar, IR.Proximity

frxnz, 2015-03-05:
@rwaldron I'm hoping to get started on this this weekend. What is the plan for deprecating the existing classes?

rwaldron, 2015-03-06:
None yet, so let's see where we're at once you're ready :)

frxnz, 2015-04-03:
Finally getting back on this. I've got the analog distance sensors refactored, moving on to what is currently `IR`/`IR.Proximity`.

Do you want to continue emitting the `motionstart` and `motionend` events? The `GP2Y0D805Z0F` sensor is the only one emitting them right now, and for a proximity class I'm not sure they make sense. Just wanted to double check before I remove them.

rwaldron, 2015-04-03:
You're right, that's an oddball—I think that has to move to IR.Motion, which means we need to refactor that class as well (which is ok, it was just a matter of time!)

frxnz, 2015-04-03:
Ok I just did a little reading up on the GP2Y0D805Z0F. Since it's more of a boolean sensor and doesn't return a distance to obstruction, it doesn't belong in the proximity class.

In that case I'll move on to Ping and Sonar. I may need to pick up a few more sensors.

rwaldron, 2015-04-03:
>  I may need to pick up a few more sensors.

I can also help test

frxnz, 2015-04-03:
> I can also help test

Ok thanks! I'll pick up any sensors that I may use and/or are not crazy expensive.

frxnz, 2015-04-29:
Closed with #688

closed 2015-04-29 by frxnz

rwaldron, 2015-04-30:
:clap: