rwaldron/johnny-five #636: Add `isAnode` to ShiftRegister constructor options object

rwaldron, 2015-02-24:
eg.

```js
new five.ShiftRegister({
  isAnode: true|false,
  pins: {
    ...
  }
});
```

See: 
https://github.com/rwaldron/johnny-five/pull/596/files#diff-bf2e64aa10778db89c9dad114b241952R171

Setting `isAnode` to `true` must invert bits sent to the register. 

cc @boneskull this is all yours

boneskull, 2015-02-24:
This seems to assume there's a seven-segment LED on the other side of the shift register... maybe a `SevenSegment` class or LED controller or something might be more appropriate?

I mean, isn't a shift register just a kind of multiplexer?  So you could use it on say, something ATtiny based, if you don't have enough pins to do what you want?

rwaldron, 2015-02-24:
> I mean, isn't a shift register just a kind of multiplexer?

Sure, but you're code is being forced to do some kind of extra work before calling the `send()` method—that seems unfortunate. I'm sure there are other cases where the data that's sent must be inverted? Let's try to identify those cases and perhaps instead of `isAnode`, we use `isInverted`?

closed 2019-01-10 by stale[bot]