rwaldron/johnny-five #638: Johnny-Five Suddenly Starts Throwing this Error

Jabbath, 2015-02-26:
I have been making a node-webkit application that uses johnny-five and it has been working great up until now. Out of the blue node-webkit always crashes with the following error

```
TypeError: Uncaught, unspecified "error" event.
    at TypeError (<anonymous>)
    at Board.EventEmitter.emit (events.js:74:15)
    at Board.eval (C:\Users\Admin\Downloads\Rebuild\node_modules\johnny-five\lib\board.js:383:12)
    at Timer.listOnTimeout [as ontimeout] (timers.js:110:15)
```

I have not changed anything involving johnny-five and even the most minimal example throws this error. This error persists if I rebuild the whole app from scratch. 

I am using Windows 7 64 bit with node-webkit 0.8.6.

It seems that other users are also having this problem, judging by [this issue](https://github.com/monteslu/bean-io/issues/7).

rwaldron, 2015-02-26:
I haven't successfully built node-serialport for node-webkit in about a year :\ Can you post the program code here? What version of Johnny-Five is being used?

Jabbath, 2015-02-26:
I am using v.0.8.44 of johnny-five. This is the most minimal example with which I can duplicate the error.
```
var five = require('johnny-five');
var board = new five.Board();
board.on('ready', function(){});
```

I have a build guide for my app that seems to build serialport every time you can find it [here](https://github.com/Jabbath/Gonzaga-ROV).

rwaldron, 2015-02-26:
Thanks! I will work on this tomorrow (appreciate your patience)

rwaldron, 2015-02-26:
Unrelated to this bug: I see that `Servo` is being used to control motors? Are those ESCs? If yes, then you can use the `ESC` class: https://github.com/rwaldron/johnny-five/wiki/ESC

rwaldron, 2015-02-26:
I've encountered a few things so far, but nothing that I can say is definitively connected to this issue: 

- This commit from 17 hours ago: https://github.com/Jabbath/Gonzaga-ROV/commit/ed06f669ed5b8e0f1856251918805e25a527328e#diff-e5e78384e0a204ddfd6c9caf2f4050dd , so I uncommented that. 

- This error when running the program: 
```
[57207:0226/100728:INFO:CONSOLE(343)] "Uncaught Error: Cannot find module '/Users/rwaldron/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.5.0/Release/node-webkit-v11-darwin-ia32/serialport.node'", source: module.js (343)
```

So I tried: 

```
rwaldron at nova in ~/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport on master*
$ nw-gyp rebuild --target=0.8.6
gyp info it worked if it ends with ok
gyp info using nw-gyp@0.12.4
gyp info using node@0.10.35 | darwin | x64
gyp info spawn python
gyp info spawn args [ '/usr/local/lib/node_modules/nw-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/rwaldron/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/nw-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/rwaldron/.nw-gyp/0.8.6/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/rwaldron/.nw-gyp/0.8.6',
gyp info spawn args   '-Dmodule_root_dir=/Users/rwaldron/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp: Undefined variable module_name in binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/nw-gyp/lib/configure.js:353:16)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Darwin 14.1.0
gyp ERR! command "node" "/usr/local/bin/nw-gyp" "rebuild" "--target=0.8.6"
gyp ERR! cwd /Users/rwaldron/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport
gyp ERR! node -v v0.10.35
gyp ERR! nw-gyp -v v0.12.4
gyp ERR! not ok
```

Then I went and found this: https://github.com/voodootikigod/node-serialport/commit/1613243d0b7dc43e3c8ab555cc55f18842b4adca#diff-21e418851ae48f9b11907837add9e39a, so reverted that locally (read: deleted these lines)—which was successful: 

```
rwaldron at nova in ~/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport on master*
$ nw-gyp rebuild --target=0.8.6
gyp info it worked if it ends with ok
gyp info using nw-gyp@0.12.4
gyp info using node@0.10.35 | darwin | x64
gyp info spawn python
gyp info spawn args [ '/usr/local/lib/node_modules/nw-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/rwaldron/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/nw-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/rwaldron/.nw-gyp/0.8.6/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/rwaldron/.nw-gyp/0.8.6',
gyp info spawn args   '-Dmodule_root_dir=/Users/rwaldron/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]

...

  SOLINK_MODULE(target) Release/serialport.node
  SOLINK_MODULE(target) Release/serialport.node: Finished
gyp info ok
```

But, then I hit this: 

```
59733:0226/102814:INFO:CONSOLE(343)] "Uncaught Error: Cannot find module '/Users/rwaldron/clonez/Gonzaga-ROV/node_modules/johnny-five/node_modules/serialport/build/serialport/v1.5.0/Release/node-webkit-v11-darwin-ia32/serialport.node'", source: module.js (343)
```

And now I'm not sure what to do next.

Jabbath, 2015-02-26:
Yep, they are esc's so I will definitely look into that class. As for your problem with serialport, I believe the issue lies in the fact that you have to use node-pre-gyp to rebuild it. Try
```
node-pre-gyp rebuild --runtime=node-webkit --target=0.8.6
```

rwaldron, 2015-02-26:
Experimenting a little...

Starting at the root of the project...

```bash
# If `serialport` currently installed...
# npm uninstall serialport
npm uninstall serialport && npm install serialport;

cd node_modules/serialport/;

# remember to edit binding.gyp

nw-gyp rebuild --target=0.8.6;

# Return to project root...
cd  ../..;

# Copy serialport.node created by nw-gyp 
cp node_modules/serialport/build/Release/serialport.node ./;

# Restore the path that node-webkit wants to find serialport.node at
mkdir -p node_modules/serialport/build/serialport/v1.5.0/Release/node-webkit-v11-darwin-ia32/

# Restore serialport.node
cp serialport.node node_modules/serialport/build/serialport/v1.5.0/Release/node-webkit-v11-darwin-ia32/;
```

main.js: 
```js
var serialport = require("serialport");

serialport.list(function(err, ports) {
  ports.forEach(function(port) {
    console.log(port);
  });
});
```

Success: 

```
[64261:0226/112125:INFO:CONSOLE(22)] ""/dev/cu.Bluetooth-Incoming-Port"", source: file:///Users/rwaldron/clonez/Gonzaga-ROV/bin/main.js (22)
[64261:0226/112125:INFO:CONSOLE(22)] ""/dev/cu.Bluetooth-Modem"", source: file:///Users/rwaldron/clonez/Gonzaga-ROV/bin/main.js (22)
[64261:0226/112125:INFO:CONSOLE(22)] ""/dev/cu.usbmodem1421"", source: file:///Users/rwaldron/clonez/Gonzaga-ROV/bin/main.js (22)
```

Then I copied the `node_modules/serialport` to `node_modules/johnny-five/node_modules/serialport`. Updated main.js: 

```js
var five = require("johnny-five");
var board = new five.Board({
  repl: false,
  debug: false
});
board.on("ready", function() {
  console.log("READY!");
});
```

Success: 

```
[64528:0226/112804:INFO:CONSOLE(14)] ""READY!"", source: file:///Users/rwaldron/clonez/Gonzaga-ROV/bin/main.js (14)
```

Restored main.js to the current master, manually applied these changes: 

```diff
var five = require('johnny-five');
- var nodecontroller = require('node-usbcontroller');
+ // var nodecontroller = require('node-usbcontroller');
var config = require('./data/attack3.json');
var gui = require('nw.gui');

- var newWin = gui.Window.open('./bin/camview.html',{position: 'center',
- width: 1300,
- height: 1024}); //Open our camera window
+ // var newWin = gui.Window.open('./bin/camview.html',{position: 'center',
+ // width: 1300,
+ // height: 1024}); //Open our camera window

- //var board = new five.Board();
+ var board = new five.Board({
+   repl: false,
+   debug: false
+ });
```

And commented out lines 67-135 because I don't have a controller handy

Running the program fails at this point: 

[image]

This is because index.html has script sources that don't exist (ie. there is no source for them in the repo and no instruction for installing): 

```html
<script src="bin/jquery-2.1.1.js"></script>
<script src="bin/jquery-ui-1.11.2/jquery-ui.js"></script>
```

So then: 

```bash
cd bin
wget https://code.jquery.com/ui/1.11.2/jquery-ui.min.js
wget https://code.jquery.com/ui/1.11.2/themes/smoothness/jquery-ui.css
wget http://code.jquery.com/jquery-2.1.3.js
cd ..
```

And updated index.html: 

```diff
- <link rel="stylesheet" href="bin/jquery-ui-1.11.2/jquery-ui.css">
+ <link rel="stylesheet" href="bin/jquery-ui.css">

...

- <script src="bin/jquery-2.1.1.js"></script>
- <script src="bin/jquery-ui-1.11.2/jquery-ui.js"></script>
+ <script src="bin/jquery-2.1.3.js"></script>
+ <script src="bin/jquery-ui.min.js"></script>
```

Now running the program works: 

```bash
rwaldron at nova in ~/clonez/Gonzaga-ROV on master*
$ node-webkit.app/Contents/MacOS/node-webkit .
2015-02-26 11:44:23.072 node-webkit[64964:12755068] Internals of CFAllocator not known; out-of-memory failures via CFAllocator will not result in termination. http://crbug.com/45650
[64964:0226/114423:ERROR:breakpad_mac.mm(234)] Breakpad initializaiton failed
2015-02-26 11:44:28.268 node-webkit Helper[64967:12755620] Internals of CFAllocator not known; out-of-memory failures via CFAllocator will not result in termination. http://crbug.com/45650
2015-02-26 11:44:28.403 node-webkit Helper[64968:12755668] Internals of CFAllocator not known; out-of-memory failures via CFAllocator will not result in termination. http://crbug.com/45650
[64964:0226/114433:INFO:CONSOLE(139)] ""board connected"", source: file:///Users/rwaldron/clonez/Gonzaga-ROV/bin/main.js (139)
```

[image]

-------

That was a lot of hacking around, but I never encountered the crash reported in the OP. I don't have a windows machine available to test with, but I'm going to ask around and see if anyone @bocoup can help me out.

rwaldron, 2015-02-26:
Sorry, I didn't see your last message until just now (I've had my previous comment in the editor for over an hour and wasn't paying attention). I will also try your suggestion.

Jabbath, 2015-02-26:
It seems like my issue may have come from some sort of system problem. I re installed windows after I noticed some other weird stuff happening, and now the program behaves as expected (also tested on another windows machine).

Thanks for all the effort, I really appreciate it!

rwaldron, 2015-02-26:
@Jabbath I'm glad everything is back to normal :)

closed 2015-02-26 by rwaldron