rwaldron/johnny-five #639: Piezo support for Ring Tone Transfer Language

Resseguie, 2015-02-28:
Should the piezo library support the [Ring Tone Transfer Language (RTTL)](http://en.wikipedia.org/wiki/Ring_Tone_Transfer_Language) as an input format?

Then we could immediately import all kinds of song collections available on the web:
e.g.: http://ez4mobile.com/nokiatone/rtttf.htm

cc @McDonnellDean @lyzadanger @julianduque @BrianGenisio

rwaldron, 2015-03-01:
Sounds exciting!

lyzadanger, 2015-03-01:
Sounds kinda curiously fun!

julianduque, 2015-03-01:
I like it!

McDonnellDean, 2015-03-23:
Sorry for the delay getting back, working on a job move (to my first node job yay!)

I think this should mostly work already, mostly.

Let me have a look over the next couple of days and see what changes are needed to get this going but I'm pretty sure we are only limited by the Peizo itself.

McDonnellDean, 2015-03-25:
Going to take a stab at this enhancement this weekend.

McDonnellDean, 2015-03-25:
PS... I need a hero by Bonny Tyler will be the test tone... ya'll know why :smile:

rwaldron, 2015-03-25:
That's a seriously emotional montage :heart:

McDonnellDean, 2015-03-25:
Absolutely!

McDonnellDean, 2015-03-29:
@lyzadanger  @rwaldron @julianduque @Resseguie @BrianGenisio 

So I think I have a proposal for this to share

_Saria's Song: Zelda:OoT -_
`"d=8 o=4 b=125: f a 4b f a 4b f a 4b e5 4d5 b c5 b g 2e a b g 2e 2p"`

The new syntax is basically RTTL+ but plain old RTTL is also supported. It breaks down like so:

`"d=8 o=4 t=125:"`    
- set the detault note duration & octave, and the tempo
- `b` can be used instead of `t` to support legacy tunes

`"f a 4b f a 4b f a 4b e5 4d5 b c5 b g 2e a b g 2e 2p"`
- notes are in the format _duration-note-octave_
- both the duration and octave are optional
- appending a note section with a `.` will play the note again at have the beat
- notes can be seperated by a `,` to support legacy tunes
- `p` is used for no-tone, duration can be specified

So thats basically what I have. Other points to be aware of

- Older tones don't sound great without some changes, I think we need a quirks mode
- The syntax is case insensitive
- All parts of RTTL are supported
- Could be standardised and used by all frameworks for piezo songs.

I pretty much wanted to throw this up here and get some feedback before I start coding
it up. If everyone is happy it should only take a 3 or so days to complete.

Resseguie, 2015-03-30:
I like it!

lyzadanger, 2015-03-30:
I think you should start workin' on it! Curious to see what "quirks mode" ends up looking like.

McDonnellDean, 2015-03-30:
@lyzadanger Gonna start on it tonight I think. Quirks mode is really only a flag that will do some standard shifting of values to ensure that older tones work. Not sure how to implement it yet though, I need to be mindful that I can't assume all tones will shift by the same about. Either way though I think we will get close enough to get RTTL tones working and have a nice input method too :D

McDonnellDean, 2015-04-07:
Ok all,

I've come across something I need some feedback on. I mentioned this in gitter already but I thought I would post here for posterity.

The issue has to do with `tone()` vs `frequency()` For clarity let me show the relationship of some notes:

- Note: A4 = Freq: 440Hz

If I pass a note's frequency to tone, `tone(440, 200)` I get an A4 note for 200ms. If I pass the same to frequency, `frequency(400,200)` I get D6'ish (1136) instead due to calculations done in `frequency()`. The interesting thing is that all song notes passed to `play` are first used to look up the frequency in  `Piezo.Notes` and then passed to `frequency()`. 

I'm trying to establish if we have a bug here or if there is a hardware reason for doing this modification against the notes.

The only thing I can think of is that we don't actually send the tone to the Piezo so I wonder if this has to do with ensuring the pulsing of the pin matches the proposed frequency.

Any feedback on this would be great. While I am making my changes I will also look at making some of this knowledge explicit within the script.

rwaldron, 2015-04-07:
This might _look_ like "RTFM", but I promise that it's not...

From https://github.com/rwaldron/johnny-five/wiki/Piezo

- **frequency(frequency, duration)** Play tone at `frequency` (in Hz) for `duration` milliseconds.

  ``` js
  piezo.frequency(587, 1000); // Play note d5 for 1 second
  ```

- **tone(frequency, duration)** Play a `tone` for `duration` ms. The `tone` value in this case is a computed duty cycle (in microseconds). This is a lower-level method than `frequency` (which does the translation from frequency to tone for you). Most of the time you likely want to use `frequency`.

I would argue for changing the docs to: 

- **tone(duty, duration)** Set a `duty` (duty cycle in microseconds) to play for `duration` (milliseconds). This is a lower-level method than `frequency` (which does the translation from frequency to tone for you). Most of the time you likely want to use `frequency(...)`.

McDonnellDean, 2015-04-07:
@rwaldron No that makes perfect sense, So I had it right almost in that we aren't setting the freq directly rather we are using it to create the duty cycle.

I'm not going to lie, I didn't even know we had a wiki section, I had assumed all the docs are in the repo proper, is there a difference between the two or is there a rule from using one over the other?

lyzadanger, 2015-04-07:
The clarification @rwaldron makes is very helpful :+1: 

The wiki vs. not-wiki has confused me in the past, too :).

Resseguie, 2015-04-07:
@rwaldron has admitted in the past that using "docs" for "generated examples" wasn't the best, but we're kinda stuck unless we are willing to break old links.

Wiki = Documentation
"/docs" = generated example files

lyzadanger, 2015-04-07:
@Resseguie Old links from external places, you mean? As in, we don't want to kill extant URLs? I'm very keen on helping with this...while I don't necessarily think the Wiki needs to stop existing, I do think finding docs could be clearer. Sign me up (in another thread, ob'ly) if there's ever a docs project!

rwaldron, 2015-04-07:
> @rwaldron has admitted in the past that using "docs" for "generated examples" wasn't the best, 

That's putting it nicely... I **hate** that I called it that. 

@lyzadanger 

> I do think finding docs could be clearer

I completely agree. Here's what we can do:

- new website in the works, I will chat with you on gitter about that if you like
- add a nice bold header + link (to wiki for now, website later) at the top of all the generated examples.

Resseguie, 2015-04-07:
>add a nice bold header + link (to wiki for now, website later) at the top of all the generated examples.

I'll add that as part of #700 or related.

McDonnellDean, 2015-04-21:
See #741 For current status of the above

scottgonzalez, 2015-06-02:
What's the status of this? Does the potential inclusion of Octavian (#809) come into play?

mcdonnelldean, 2015-06-03:
@scottgonzalez It is more or less ready, some minor bugs, needs more tests but the bulk of the work is finished. #809 shouldn't affect the public API but I would imagine we would want to get Octavian in and tested before calling this done.

Is there any specific use cases you are thinking about?

scottgonzalez, 2015-06-03:
Nope. I didn't dig into either one, just wanted to make sure that it was taken into account if there was going to be any overlap.

dtex, 2018-01-11:
Hi @Resseguie ,

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this feature yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for the RTTL (and RTTL+) there.

closed 2018-01-11 by dtex