rwaldron/johnny-five #641: Motor docs pin reference

Jackbennett, 2015-03-01:
Hello, was just trying to get my motor reverse from reading [this example](https://github.com/rwaldron/johnny-five/blob/master/docs/motor-hbridge.md) and the [motor class reference](https://github.com/rwaldron/johnny-five/wiki/Motor)

I'm using an Uno with a L293D h-Bridge out the starter kit and it took me a while and someone else's example to try using `cdir` in my pins object to get the motor to reverse. I think the docs could make it a bit clearer somewhere that `dir` and `cdir` might be used together. I was trying to use `brake` at first.

dtex, 2015-03-01:
Hi @Jackbennett ,

CDIR would only be necessary if your wiring is different than the diagram in the example. 

Do you have a wiring diagram or a picture you could share so I can make sure what is in my head matches what you've wired up?

Jackbennett, 2015-03-01:
I was using this diagram without the button as a reference to get it to work;

http://api.ning.com/files/jFhR10F2nKDm5cJ0p2jcpgHpKd4hNPz01QgMS8OF6QUh82aIgvr-dp1uzQSopb*PQgSlhFEgj5ALhek3X56Zq6xDZuw-9vwD/l293dmontagem.jpeg
[image]

Here's my actual diagram I just made;

[image]

Just to clarify, I don't think the example is wrong, but I can't see any clear mention that doing it with the 3 pins that made more sense to me requires the cdir option. I'm bringing it up because the L293D chip is in the arduino starter kit so someone else new might run into it.

I was looking to see if `dir` could take an array. i.e `dir: [8, 12]`

dtex, 2015-03-02:
Cool, thanks for the diagram.

I feel kinda weird providing a 3 pin example for the H-Bridge since it can be done with 2. I don't want to send people down a less efficient path. However, I like to be accommodating and I certainly don't know everything there is to know about H-Bridge's so please help me understand the case for doing it with three.

Jackbennett, 2015-03-02:
This is the first thing I've made so my experience might not be the level you intend this library for  but I'll explain my reasoning for you to judge.

L293D chip pinout for reference, pin numbers form here refer to this diagram not an arduino.
[image]

The 3-way as wired I can see that;
- Pin 2 and pin 7 take a boolean input for direction on or off.
- Pin 1 on PWM then drives that whole side at the given speed.
- The enable pin is not always on
- All the enable/input voltage is from the Arduino 5v, including the logic supply pin 16 from the 5v
- Pin 8 supply voltage is clearly form the 9v

From the [datasheet](http://arduino.cc/documents/datasheets/L293D.pdf) page 2 "Absolute Maximum Ratings" Vi and Ven (input and enable voltage) list a max of 7v. I didn't want to put the 9v battery to that with pin 1. 

Just logically this way I could mentally step through what was going to happen.
1. Set the direction
2. Wind the motor

Now, Maybe I'm wrong about this. Or perhaps I should PR this as a new example to demonstrate the H-bridge before going on to cover the 2pin method. I still need to learn how just pin7 being pwm can drive the direction.

dtex, 2015-03-02:
> I still need to learn how just pin7 being pwm can drive the direction.

The key is the difference in the voltage being applied to the two poles of the motor. With pin 2 on the H-Bridge being fed a digital signal and pin 7 on the H-Bridge being fed an analog signal:

Pin 2 | Pin 7 | Difference | Result
------------ | ------------- | ------------- | -------------
Low | 0 | **0** | Stopped
Low | 64 | **+64** | Forward half speed
Low | 128 | **+128** | Forward half speed
Low | 255 | **+255** | Forward full speed
High | 0 | **-255** | Reverse full speed
High | 64 | **-191** | Reverse 3/4 speed
High | 128 | **-128** | Reverse 1/2 speed
High | 255 | **0** | Stopped

dtex, 2015-12-11:
Hi @Jackbennett, did everything work out? Can we close the issue?

Jackbennett, 2015-12-11:
I thought I might just write up an extra example for a PR but never got to it.

Excellent explanation thank you for your help.

closed 2015-12-11 by Jackbennett