rwaldron/johnny-five #642: Is there a way to avoid the REPL in Johnny-Five?

scazan, 2015-03-02:
I am using Johnny-Five to build some software for interacting with electronics and don't need the interactive console to be present (in fact it seems to be limiting me in a lot of ways). In particular, I'm having trouble detecting a SIGINT once Johnny-Five is within its own REPL.

I've been searching all over and cannot find anyone mentioning this.

BrianGenisio, 2015-03-02:
It appears so.  Have you tried this?

```js
var board = new five.Board({ repl: false });
```

Documented on the [Board Wiki](https://github.com/rwaldron/johnny-five/wiki/Board)

closed 2015-03-02 by rwaldron

alexcroox, 2015-10-29:
On the topic of SIGINT, I've noticed not all my shutdown proceedures get time to execute before J5 fires process.exit(1);. The last console log I get is "Board closing". Is there anyway to stop J5 closing the process on my behalf when I want to do other things before fully exiting.

rwaldron, 2015-10-29:
@alexcroox IIRC: `sigint: false`

alexcroox, 2015-10-30:
That works perfectly thank you! Do I need to close the board manually myself or is safe to just do process.exit(1);

rwaldron, 2015-10-30:
That just prevents johnny-five from registering it's own sigint handler, which in turn prevents the call to `process.exit(0)`