rwaldron/johnny-five #645: Error in image button.png

bennysce, 2015-03-07:
Hi, noticed on the lasercat workshop that the button image referred to at https://github.com/rwaldron/johnny-five/blob/master/docs/button.md is incorrect. I don't think that red wire is correctly placed on the board for it to work. The red wire has to be set in the same column as the leg for the button.

See johnny-five/docs/breadboard/button.png.

aramperez, 2015-03-08:
I'm not that familiar with johnny-five yet, but if the code states "button = new five.Button(8);", shouldn't the yellow wire go to pin 8 on the Arduino instead of pin 2?

rwaldron, 2015-03-09:
> I don't think that red wire is correctly placed on the board for it to work. 

You are correct :)

> I'm not that familiar with johnny-five yet, but if the code states "button = new five.Button(8);", shouldn't the yellow wire go to pin 8 on the Arduino instead of pin 2?

Also correct :)

Thanks for the bug reports—I'll have a fix for this asap.

closed 2015-03-09 by rwaldron

Resseguie, 2015-03-09:
@rwaldron FYI, you might just b e able to use the FZZ I did for your presentation a while back but move to pin 8.

https://www.dropbox.com/s/j8jkks2ojx4q8cz/button8.png?dl=0
https://www.dropbox.com/s/ckuk04d460rciqb/button8.fzz?dl=0

Resseguie, 2015-03-09:
Never mind, you beat me to the update.