rwaldron/johnny-five #646: Zumo IR Array not working with Johnny Five

den2k, 2015-03-15:
I've been having great success getting Johnny Five to work with the Zumo bot.  Motor control through wireless works great with keyboard control.

I'm now trying to get the attached IR Array to work.

The challenge I have is running the following code.

    var five = require("johnny-five");
    var board = new five.Board({
    	port: "com3"
    });

    var stdin = process.stdin;
    stdin.setRawMode(true);
    stdin.resume();

    board.on("ready", function() {

      // Create a new `reflectance` hardware instance.
      var eyes = new five.IR.Reflect.Array({
        emitter: 2,
        pins: [4, "A3", "A0", "A2", 5], // cannot use 11, and lots of pins are already in use?
      //    emitter: 2,  // Zumo Reflectance Sensor Array Emitter Pin for Arduino Uno
      //    pins: ["4", "A3", "11", "A0", "A2", "5"],	// Zumo Reflectance Sensor Array
      });

      var calibrating = true;
  
      eyes.calibrateUntil(function () {return !calibrating; });
      console.log("Press the spacebar to end calibration...");
  

      stdin.on("keypress", function(chunk, key) {
      if (!key || key.name !== 'space') return;
        
        calibrating = false;
      });
 
      eyes.on("line", function(err, line) {
        console.log(line);
      });
  
      eyes.enable();
    });

The issue is the various pins are already in use.  

pin: 2 is already in use
pin: 4 is already in use
pin: 3 is already in use
pin: 0 is already in use
pin: 2 is already in use
pin: 5 is already in use

and calibration doesn't work properly.  I'm new, so it could be just entirely my issue.  I would appreciate any direction.

BrianGenisio, 2015-03-15:
A couple of things related to this:

The `IR.Reflect.Array` currently only supports analog sensors.  The Zumo array is a digital sensor, which uses pulse timing to determine the values.  Instead, if you want to use a reflectance array in Johnny-Five, you will need to use an [analog array](https://www.pololu.com/product/960).  

Because of this, you need to use analog pins only (A0, A1, A2, etc).  In your example, you are mixing digital and analog pins.  I don't know if this is why you are getting the "already in use" error, but it could be because J5 is trying to set the digital pins as analog?

den2k, 2015-03-15:
I didn't realize this made a difference. Good to know and seems obvious now. Luckily I have an analog version of their IR array. I'll try that when I get a chance and see if it makes a difference. I'll also just try with just the analog pins on the current array and see what I get.

BrianGenisio, 2015-03-16:
Cool.  Glad to know you have an analog one lying around.  I can assure you, however, that if you connect a digital array up to analog pins, the driver won't work.  The encoding is completely different.  As far as I understand, the Zumo array only comes in digital.  The Pololu sensor comes in digital or analog.

den2k, 2015-03-25:
I hooked up a QTR-8A array to the Zumo bot and I now have readings, but I don't think it is calibrating correctly with the following code.  I have 5 sensors hooked up and I should be seeing 0 - 4001 I believe.  Not sure what the 2000 base value means in my case.  I also continue to see Pin in use when I run this code. Hmmm?

var five = require("johnny-five");
var board = new five.Board({
	port: "com3"
});

board.on("ready", function() {

  var calibrating = true;

  var eyes = new five.IR.Reflect.Array({
    emitter: 13,
    pins: ["A0","A1","A2","A3","A4"]
  });
  
  // calibrate for two seconds
  eyes.calibrateUntil(function() { return !calibrating; });
  setTimeout(function() { calibrating = false; }, 2000); 

  eyes.enable();

  eyes.on('line', function(err, lineValue) {
    console.log( "Line Position: ", lineValue);
  });
  
});

-------------------------

pin: 0 is already in use
pin: 1 is already in use
pin: 2 is already in use
pin: 3 is already in use
pin: 4 is already in use
Line Positions: 0
Line Positions: 2500
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2000
Line Positions: 2013
Line Positions: 2010
etc.

BrianGenisio, 2015-03-25:
While calibrating, you need to make sure that all sensors see both extremes (white and black).  So in the 2 second window you created, wave the sensor at the height you expect over white and black.

2 seconds is a bit short to do it well.  I like to take a keypress input instead, and then serialize the calibration data to a file.  If the file exists, deserialize it and pass it to the component (`loadCalibration`).

I keep meaning to change this sensor to have an `autoCalibrate: true` option which will calibrate as it drives.  It would simplify the calibration process, but it would make for a messier start for a line following bot. 

But you are correct.  If everything is working properly, 5 sensors would be between 0 and 4001 for the "line" event.  It really just aggregates the "calibratedData" array into a single number which tells you where on the array the data is.  The "calibratedData" event will give you normalized data on the 5 sensors.  The "data" event will give you raw values on the 5 sensors.  

At any point in time, you also have access to the `values` property with is an array of calibrated data at the last reading and the `raw` property which is an array of raw data at the last reading.

If your raw values are changing how you'd expect as you swipe it over the line, then you know the sensor is working and wired up properly.

dtex, 2018-01-08:
@den2k Touching base. Did @BrianGenisio's comment give you everything you need? Wondering if I can close this issue.

dtex, 2018-01-16:
@den2k I'm going to close this since we haven't heard back. Feel free to re-open if necessary.

closed 2018-01-16 by dtex