rwaldron/johnny-five #649: How to handle reconnects?

fsamir, 2015-03-15:
Hi there,
what is the right way to safely handle when the user pulls off the USB cable?
Also, how to auto reconnect once the USB is connected again?

The blink example crashes when the USB cable gets pulled off.

Thanks,
Franklin

rwaldron, 2015-03-16:
Unfortunately, it's not possible to maintain state integrity when the serial line is disconnected. Once the USB is plugged back in, StandardFirmata will reset on the board.

closed 2015-03-16 by rwaldron

rexpie, 2016-03-18:
Okay, I agree that state will be out of sync. Is there a way for me to reset the whole johnny-five/firmata/serialport stack without restarting the whole node process? 

I tried to call `board.io.sp.close()` then tried to use a new five.Board(), but that gives me warnings about pins being in use when I tried to re-create the pins.

Another bummer on windows is that the serialport module does not report on usb cable disconnection when I unplug it, leaving me not many choices but to use timeout and check firmata version at intervals.

rwaldron, 2016-03-23:
> Is there a way for me to reset the whole johnny-five/firmata/serialport stack without restarting the whole node process?

No. If you need to be able to restart, I suggest controlling the program as a child process. 

> I tried to call board.io.sp.close() then tried to use a new five.Board(), but that gives me warnings about pins being in use when I tried to re-create the pins.

This appears to be working as expected. 

> Another bummer on windows is that the serialport module does not report on usb cable disconnection when I unplug 

I recommend filing an issue with node-serialport :)

rexpie, 2016-03-31:
Okay many thanks again. I resorted to use pm2 with keep alive messages to the board. :)

mircobabini, 2018-07-03:
@rexpie ...how? code example?