rwaldron/johnny-five #655: Error Device or firmware error

brunocascio, 2015-03-19:
Hi!

I have an arduino Mega 2560.

I run Kali Linux (Debian 7) and when run, i get:

```
root@kali:~/Documents/Arduino/test# node index.js 
1426735984300 Device(s) /dev/ttyACM0 
1426735984310 Connected /dev/ttyACM0 
1426735994311 Device or Firmware Error A timeout occurred while connecting to the Board. 
Please check that you've properly flashed the board with the correct firmware. 
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: A timeout occurred while connecting to the Board.
    at Board.<anonymous> (/root/Documents/Arduino/test/node_modules/johnny-five/lib/board.js:383:26)
    at Timer.listOnTimeout (timers.js:110:15)

```

I don't understand the problem...

mapaiva, 2015-03-19:
@brunocascio Have you uploaded StandardFirmata to your arduino?

rwaldron, 2015-03-19:
@brunocascio as @mapaiva suggested, please make sure that StandardFirmata has been uploaded to your board. Here are instructions: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

brunocascio, 2015-03-19:
Works! thanks @rwaldron @mapaiva !

closed 2015-03-19 by brunocascio