rwaldron/johnny-five #656: Start event is called without delay. Confusing to beginners

martinheidegger, 2015-03-19:
I just had a question at the NodeSchool Osaka where one member tried to run this:

```JavaScript
motor.start()
motor.on("start", function () {
   // do something
})
```

A not quite uncommon thing to expect to work. Since a event system is expected to have _a_ delay anyways. However johnny-five [calls the method directly](https://github.com/rwaldron/johnny-five/blob/35ab892a17bf53765720b350b32efea2da181c1a/lib/motor.js#L564) so it has to look like this:

```JavaScript
motor.on("start", function () {
   // do something
})
motor.start()
```

dtex, 2015-03-19:
It is not clear to me how this could be changed. Even if we were doing something asynchronously and could emit the event after I/O to the board we would still be at risk of a race condition.

fivdi, 2015-03-19:
Would replacing [this](https://github.com/rwaldron/johnny-five/blob/master/lib/motor.js#L563-L565):
```javascript
if (speed > 0) {
  this.emit("start", null, new Date());
}
```
with this:
```javascript
var tick = global.setImmediate || process.nextTick;
...
if (speed > 0) {
  tick(function () {
    this.emit("start", null, new Date());
  });
}
```
not improve things a lot?

dtex, 2015-03-19:
Hmmm, I wonder why we are not using that in more places. I even see where I commented it out in Servo though I have no memory of why. Perhaps there was an issue with Travis and our fake timers? We aren't even looking for that event in tests so that probably wasn't it. I'm going to dig through the history of Servo and Motor and see if I can find the reason.

martinheidegger, 2015-03-22:
Looks like a good start to me. Its always a bit scary to see changes without unit tests though :)

dtex, 2015-03-22:
So unit tests on the event emitters? Easy enough but different issue #663. Closing this.

closed 2015-03-22 by dtex