rwaldron/johnny-five #666: Servo offset bug

dtex, 2015-03-24:
There is a bug in servo that I wanted to document in case anybody else has struggled with it. It appears when a servo opts object has both a range and offset property and the problem is compounded when a servo is inverted.

The offset property applies a transform to the local coordinate space of the servo. We are currently enforcing our range limits after the offset has been applied. This means that the range is being applied on the local coordinate space. It should be applied on the global coordinate space.

Using this example of a coxa joint in global coordinate space:

````
var coxa = new Five.Servo({offset: -90, range: [155, 205]});
````

[image]
black values are global, red values are local

When the user calls a value that is outside the local range, but within the global range.

````
coxa.to(180);
// Expected behavior is to set servo to 90°
// Actual behavior is servo goes to 65°
````

If we configure our servo like this:

````
var coxa = new Five.Servo({offset: -90, range: [65, 115]});
````

when the user calls a value that is outside the local range, but within the global range.

````
coxa.to(180);
// Expected behavior is to set servo to 90°
// Actual behavior is servo goes to 115°
````

I was worried the fix might be a breaking change, but the only way this would have ever worked at all for a user is if they opened up their range properties to the point where both their local and global coordinate values were allowed:

````
var coxa = new Five.Servo({offset: -90, range: [65, 205]});
````

**But wait, it gets worse**

It is also a problem when a servo is inverted and this one is going to be tougher to fix. Assume the servo in our drawing is mounted face down so that our servo's local coordinate space is inverted and our servo is configured like this:

````
var coxa = new Five.Servo({offset: -90, invert:true, range: [65, 205]});
````

````
coxa.to(160);
// Expected behavior is to set servo to 90°
// Actual behavior is servo goes to 200°
````

The inversion is base on the global coordinate space range instead of the local coordinate space range.

This offset fix is easy, but I'm still noodling over the ```invert: true``` solution. A PR is coming soon.

rwaldron, 2015-03-24:
Nice find and even better documentation :)

dtex, 2015-04-03:
Fixed wtih 29783f8

closed 2015-04-03 by dtex