rwaldron/johnny-five #669: IS_TEST_MODE could be nicer

boneskull, 2015-03-24:
`IS_TEST_MODE` is a global, set throughout several files in `test/`, but most importantly (I think) `test/bootstrap.js`.  

This is all well and good if you are running the tests via Grunt.  However, if you wish to run the Nodeunit tests directly via an interactive debugger--thinking JetBrains' debugger--there's no way to specify the order in which the test files should run.  

That's a limitation of my debugger, however, this is an edge case.  In a perfect world, the test files *should* run in any order.  Perhaps Nodeunit should support something like Mocha's `--require` which allows you to first require a "fixture" module, like `bootstrap.js`, before tests begin; then, all of the `purge()` functions could live in the fixture module).  Maybe we should switch to Mocha, but that's another can of worms.  :smile:

I haven't touched Java since 1.2--and I have my own test framework to worry about instead of Nodeunit--so the quickest fix for me is to modify J5 to support an environment variable instead.

**Summary of proposed changes:**

- Anywhere `IS_TEST_MODE` is retrieved, `process.env.IS_TEST_MODE` is retrieved instead
- Anywhere `IS_TEST_MODE` is set, `process.env.IS_TEST_MODE` is set instead
- `Gruntfile.js` explicitly sets `process.env.IS_TEST_MODE = true;`
- `test/bootstrap.js` and references are removed
- `IS_TEST_MODE` is removed from various JSHint configs

Then, you can use whatever debugger you want, as long as you can set an environment variable in some manner.  This is a more typical solution than setting a property on `global`--I think for good reason.

> *Note: A workaround exists, which is to debug `node grunt` itself, but that bypasses Jetbrains' Nodeunit reporter, which is a nicer interface.*

rwaldron, 2015-03-24:
On the whole, I'm not really opposed to this at all, but...

> Gruntfile.js explicitly sets process.env.IS_TEST_MODE = true;

Doesn't this mean that Jetbrains still has to run grunt?

boneskull, 2015-03-24:
@rwaldron No, because I can set env vars within my IDE.

rwaldron, 2015-03-25:
So, you're going to set it explicitly, but everyone else will get via `grunt`?

closed 2015-03-30 by rwaldron

boneskull, 2015-03-31:
@rwaldron yeah, that's the idea.  just gives it some flexibility at little/no cost

commit 15f90e86 (2015-03-30): use environment to declare test mode flag; resolves #669

- Relevant files check `process.env.IS_TEST_MODE` instead of a global prop
- The value of `IS_TEST_MODE` is assumed to be a boolean
- Individual suites do not set `IS_TEST_MODE`
- `bootstrap.js` is no longer necessary and has been removed
- `Gruntfile.js` sets the `IS_TEST_MODE` flag; as long as you run the tests via `npm test` or `grunt`, your environment will be correct.  Otherwise, you must manually set the `IS_TEST_MODE` environment variable.
- simplified test targets within `Gruntfile.js` with globs
- remove `bootstrap.js` referenc [... cut]
commit ea64d943 (2015-03-30): Merge branch 'issue/669' of https://github.com/boneskull/johnny-five into boneskull-issue/669

# By Christopher Hiller
# Via Christopher Hiller
* 'issue/669' of https://github.com/boneskull/johnny-five:
  use environment to declare test mode flag; resolves #669

Conflicts:
	lib/esc.js