rwaldron/johnny-five #674: Ping module not working + Memory leak error

ecdeveloper, 2015-03-25:
Here is a very simple code:

```
var five = require("johnny-five");
var board = new five.Board();
board.on("ready", function() {
	var ping = new five.Ping(7);
	ping.on("change", function() {
		console.log("Object is " + this.in + " inches away");
	});
});
```

For some reason nothing is outputted to the stdout. The more so, after 10-15 seconds it throws an error: 
```
(node) warning: possible EventEmitter memory leak detected. 101 listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at addListener (events.js:160:15)
    at once (events.js:185:8)
    at Board.pulseIn (/Users/evghenic/dev/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1265:8)
    at Ping.<anonymous> (/Users/evghenic/dev/node_modules/johnny-five/lib/ping.js:45:13)
    at wrapper [as _onTimeout] (timers.js:261:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:112:15)
```

If I remove the Ping related code - the error doesn't occur. 

I followed the Ping setup guide. Uploaded the custom Firmata sketch. 

I also tested the device/circuit with an Arduino application - it works fine. 

Any ideas what may be wrong?

rwaldron, 2015-03-26:
Can you send a picture of your circuit wiring?

ecdeveloper, 2015-03-26:
Red is 5V on Arduino, and + on the board
Blue is GND and -

![ping circuit]
(http://oi62.tinypic.com/mu8yo6.jpg)

cybertiti, 2015-03-26:
Hi, I had the same issue before changing to the firmata modified sketch.
With the good firmata version, everything is working.

ecdeveloper, 2015-03-26:
@cybertiti which method for changing Firmata have you used? I actually used both (cloning the jgautier/arduino-1.git repo, and also copy/pasting the Firmata code into the Arduino IDE).

rwaldron, 2015-03-26:
@ecdeveloper thanks for the clear image :heart:. I will assemble your circuit today and try to reproduce :)

ecdeveloper, 2015-03-26:
@rwaldron Thanks! Looking forward to see your results :)

rwaldron, 2015-03-26:
@ecdeveloper (unrelated to report) I notice that the board has "JY-MCU" printed on it—does this board have built-in Bluetooth? (I've never seen a board with that printed on it)

ecdeveloper, 2015-03-26:
Wow. Just noticed that too :D I don't really know, when I ordered it I was sure it's a regular UNO without any extra built-in features.

rwaldron, 2015-03-26:
Interesting—would you mind taking a few pictures from different angles and of the back side? If not, that's ok :)

ecdeveloper, 2015-03-26:
@rwaldron https://www.dropbox.com/s/88t1066vahuztxt/arduino%20uno.zip?dl=0

rwaldron, 2015-03-26:
You're the best :)

ecdeveloper, 2015-03-26:
:smile:

cybertiti, 2015-03-26:
@ecdeveloper I used the cloning git version, but you need to check is the firmata directory used by IDE is the right one by searching pulse_in in your sketch. The first time i did it, i uploaded the standard firmata because IDE was loading from the wrong directory.

ecdeveloper, 2015-03-26:
@cybertiti Hm, probably you're right. I remember when I tried cloning the repo, I didn't find any "pulse_in" matches in sketch. Will double check tonight, thanks.

rwaldron, 2015-03-26:
I just used the "copy/paste" method and cannot reproduce the issue. Here is the circuit I assembled: 

[image]

[image]

And here's a video demo of the circuit controlled by the program in `eg/ping.js`: 

https://www.youtube.com/watch?v=puDJJWomqzc

-----

The error you're seeing is usually the result of a wiring issue that causes the board to stop responding. A response is necessary to clear the event created by Firmata.js's `pulseIn` method ( `once("pulse-in-" + pin, ...)` ). 

Can you try connecting directly from the breadboard to the Ground and 5V as shown in my circuit?

ecdeveloper, 2015-03-27:
@rwaldron Thanks!!! Connecting it directly solved the problem! :)

closed 2015-03-27 by ecdeveloper

soundanalogous, 2015-03-27:
pulse_in (in Arduino / c) has a while loop that is blocking until the pulse has finished. This will create significant delays if you are trying to read other sensors and/or control actuators at the same time. Data will still buffer through the UART as best it can but you risk loss of data by using pulse_in. This is why I have never included the routine in StandardFirmata. Once someone creates an interrupt driven pulse_in routine that works across the majority of Arduino boards, I will add an implementation... likely to ConfigurableFirmata and StandardFirmata as well if the implementation it has a low memory footprint and does not interfere with existing timers.

ecdeveloper, 2015-03-27:
@rwaldron I will try to connect it back how it has been connected before, will let you know either it works or not. 

@soundanalogous It seems related to this issue - https://github.com/rwaldron/johnny-five/issues/675

ecdeveloper, 2015-03-27:
@rwaldron Now it works with my previous circuit. Magically :D

marvindebruin, 2015-04-15:
Ive tried connecting it directly and indirectly and both give the memory leak error. Maybe closing this issue was a bit premature?

 warning: possible EventEmitter memory leak detected. 101 listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at EventEmitter.addListener (events.js:160:15)
    at EventEmitter.once (events.js:185:8)
    at Board.pulseIn (/Volumes/Work HDD/Repositories/arduino/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:1265:8)
    at Ping.<anonymous> (/Volumes/Work HDD/Repositories/arduino/node_modules/johnny-five/lib/ping.js:45:13)
    at wrapper [as _onTimeout] (timers.js:252:14)
    at Timer.listOnTimeout [as ontimeout] (timers.js:110:15)

ecdeveloper, 2015-04-15:
@marvindebruin In my case the problem was because of insufficient power. I was using usb power, and I had multiple things connected to my board.

rwaldron, 2015-04-15:
>  Maybe closing this issue was a bit premature?

No, because the issue isn't software related :(

marvindebruin, 2015-04-15:
Ill add an external power supply to the mix and get back at it tomorrow. Hope this solves it.

@rwaldron I get it now :)

rwaldron, 2015-04-15:
Cool, let us all know how it goes :)

marvindebruin, 2015-04-16:
So I added an external power source (5v adapter hooked up to a second arduino) and turned it on. Still no success. Ive measured the Voltage and it gives 5V to the sensor. so that works. Ive tried the sensor in this conf with the basic ping script and that works.
[image]
[image]

rwaldron, 2015-04-16:
> So I added an external power source (5v adapter hooked up to a second arduino) 

Did you share a ground line across the two Arduinos? This really should not be necessary at all. Please try with wiring exactly as shown here: https://github.com/rwaldron/johnny-five/issues/674#issuecomment-86586234

rwaldron, 2015-04-16:
[image]

marvindebruin, 2015-04-16:
Hey Rwaldron, this made no difference. Could it be Im using a different firmata / where can I get the proper one if not from the examples. Ill try that first just to rule it out.

rwaldron, 2015-04-16:
> Could it be I'm using a different firmata

Yes, it very well could be. Are you using the special Ping-specific Firmata that's provided in the docs? https://github.com/rwaldron/johnny-five/wiki/Ping#setup

bbence84, 2015-04-30:
I have a slightly different issue / question, sorry for hijacking the thread. :) What I am trying to do is to check periodically the sensor from a node red flow. The sensor returns the correct value just fine in a vanilla node.js app, but I cannot get it to work in node red (http://nodered.org/docs/hardware/arduino.html). It returns zero. How do I connect the onchange callback in a node red flow? I tried the following which did not work:
var ping = new five.Ping(7);
msg.payload = ping.cm;

johnny five and the board is supposed to be already initialized according to the example I linked. But that is just about blinking a red, there's no callback required there... Thanks!

rwaldron, 2015-04-30:
Try this: 

```js
var ping = new five.Ping(7);

ping.on("change", function() {
  msg.payload = this.cm;
});
```

bbence84, 2015-04-30:
Thanks, but this does not seem to work.
I think the reason is that node-red isn't supposed to work in this callback fashion, Or does it? How would you use the johnny five pushbutton object in node-red?

rwaldron, 2015-05-01:
Sounds like a bug in node-red, because all of Johnny-Five's sensor/input component classes are emitters.