rwaldron/johnny-five #675: Firmata modifed for ping, other sensor not working anymore

cybertiti, 2015-03-26:
Hi,
My sensor "new five.Sensor({pin: "A5", freq: 60000})" is not working anymore since I changed for the modified firmata needed for using the "new five.Ping(0)".
Do i need to change something for Sensor library to work with modifed firmata ?
Best regards,
Thierry

rwaldron, 2015-03-26:
Can you provide your complete code example? A picture of your circuit would also be helpful—thanks in advance!

cybertiti, 2015-03-26:
Sure, here is the last part of the code, the complete one is a bit long :
if (args.indexOf("noArduino") == -1) {
  board = new five.Board();
  board.on("ready", function() {
    var controller = process.argv[2] || "GP2Y0A41SK0F";
    motorL = new five.Motor({pins: {pwm: 5, dir: 4}});
    motorR = new five.Motor({pins: {pwm: 6, dir: 7}});

    tilt = new five.Servo({pin: 10, range: [0, 60], startAt: 40, center: false});
    pan = new five.Servo({pin: 9, range: [0, 180], startAt: 90, center: true});
    led = new five.Led({pin: 13, type: "PWM"});
    light = new five.Led({pin: 12});
    laser = new five.Led({pin: 14});
    alarm = new five.Led({pin: 11, type: "PWM"});
    voltmetre = new five.Sensor({pin: "A5", freq: 60000});
    pingL = new five.Ping({pin: 0, freq: 300});
    pingR = new five.Ping({pin: 1, freq: 300});
    proxiFR = new five.IR.Distance({pin: "A1", controller: controller});
    proxiFL = new five.IR.Distance({pin: "A2", controller: controller});
    proxiRR = new five.IR.Distance({pin: "A3", controller: controller});
    proxiRL = new five.IR.Distance({pin: "A4", controller: controller});

    voltmetre.on("data", function(err, res) {
      console.log(res + " pin A5");
      res = res * 5 / 1024;
      res = res / (24930 / (47530 + 24930));
      res = res - 0.10;
      volt = res.toFixed(2);
      console.log(volt + " Volt");
    });

    pingL.on("data", function(err, value) {
      console.log("Dist Left " + this.cm + " cm");
    });
    pingR.on("data", function(err, value) {
      console.log("Dist Right " + this.cm + " cm");
    });

    proxiFR.on("change", function() {
      console.log("IR FR cm: ", this.cm);
    });
    proxiFL.on("change", function() {
      console.log("IR FL cm: ", this.cm);
    });
    proxiRL.on("change", function() {
      console.log("IR RL cm: ", this.cm);
    });
    proxiRR.on("change", function() {
      console.log("IR RR cm: ", this.cm);
    });

    serverStatus.hasArduino = true;
  });
}

cybertiti, 2015-03-26:
[image]

cybertiti, 2015-03-26:
[image]
And the front view with the sensors

cybertiti, 2015-03-26:
The Sensor on A5 is for voltage sensing, and was working perfectly before i changed the firmata with the modifed firmata for Ping library

rwaldron, 2015-03-26:
> The Sensor on A5 is for voltage sensing

Can you provide a link? I will likely need to purchase whatever you're using

Before we embark on solving this issue, all I ask is that you remain patient with me :) In some cases we'll be at the mercy of shipping times if I have to order components.

cybertiti, 2015-03-26:
[image]
You need 2 resistor, in my case i used a 47 Kohm for R1 and a 25 Kohm for R2.
Resistor value can be change, these one are for a Vin up to 14 Volt

rwaldron, 2015-03-26:
Oh great, that's very simple. I have just ordered the Romeo v2.2

cybertiti, 2015-03-26:
Oh great, Romeo v2 is really a nice board. If you zoom on my first picture, you can see the resistor plugged in the A5 blue analog input, the Vin is plugged before the DC converter, label VIN on the board. It's really easy to build, and working perfectly to monitor battery voltage.

rwaldron, 2015-03-30:
Just checking in—I haven't received the Romeo yet, but will update when I do.

cybertiti, 2015-03-31:
Hi Rick, i hope your romeo will arrive soon !
Just an update, other library like LED are not working correctly neither with modified firmata.
I revert back to the original firmata and everything is working fine again, except US and IR proximity sensor for sure.

rwaldron, 2015-03-31:
Quick question, did you select Arduino Leonardo when you uploaded the Ping Firmata?

rwaldron, 2015-03-31:
Sorry I missed this before now: the Ping instances are being set up with pin 0 and 1, which is not allowed because those are the RX/TX lines being used by the Pi to talk to the Romeo.

rwaldron, 2015-03-31:
An update. 

I received the Romeo yesterday afternoon and today I assembled the following: 

[image]

- A1: GP2Y0A21YK \*
- A2: GP2Y0A21YK
- A3: GP2Y0A21YK
- A4: GP2Y0A21YK
- A5: Slide Potentiometer as a substitute for the voltage level sensor, I wanted to be able to change the value at will. 
- D7: Ping
- D8: Ping

The following program was used to test. It was adapted from the code provided. 

Relevant changes: 
- Changed Ping pins to 7 & 8 from 0 & 1

Irrelevant changes: 
- Updated IR controller to GP2Y0A21YK \*
- Minor tweak in the `voltmetre` change handler to use `this.value` instead of the argument

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
  var voltmetre = new five.Sensor({pin: "A5"});
  var pingL = new five.Ping({pin: 7, freq: 300});
  var pingR = new five.Ping({pin: 8, freq: 300});
  var proxiFR = new five.IR.Distance({pin: "A1", controller: "GP2Y0A21YK"});
  var proxiFL = new five.IR.Distance({pin: "A2", controller: "GP2Y0A21YK"});
  var proxiRR = new five.IR.Distance({pin: "A3", controller: "GP2Y0A21YK"});
  var proxiRL = new five.IR.Distance({pin: "A4", controller: "GP2Y0A21YK"});

  voltmetre.on("change", function() {
    var res = this.value;
    console.log(res + " pin A5");
    res = res * 5 / 1024;
    res = res / (24930 / (47530 + 24930));
    res = res - 0.10;

    var volt = res.toFixed(2);
    console.log(volt + " Volt");
  });

  pingL.on("data", function(err, value) {
    console.log("Dist Left " + this.cm + " cm");
  });
  pingR.on("data", function(err, value) {
    console.log("Dist Right " + this.cm + " cm");
  });
  proxiFR.on("change", function() {
    console.log("IR FR cm: ", this.cm);
  });
  proxiFL.on("change", function() {
    console.log("IR FL cm: ", this.cm);
  });
  proxiRL.on("change", function() {
    console.log("IR RL cm: ", this.cm);
  });
  proxiRR.on("change", function() {
    console.log("IR RR cm: ", this.cm);
  });
});
```

I ran the program several times, with several logging configurations (limiting logging to certain sensors at a time). All runs behaved as expected: 

- The potentiometer reported changes for variation in voltage resistance
- The IR sensors reported readings that were consistent with changes to the physical space around them (ie. If I moved my hand across them, the readings reflected this change).
- The Ping sensors reported readings that were consistent with changes to physical space around them (ie. If I moved my hand across them, the readings reflected this change).

I used a centimeter ruler to confirm reported distances on "randomly" selected sensors on each run. 

\* GP2Y0A21YK is similar enough to the GP2Y0A41SK0F, it shouldn't make a difference.

cybertiti, 2015-04-06:
Hi Rick,
I'm glad to see you've got everything working, maybe my issue is because i used the input 0 and 1. I'm going to test with another input.
Which firmata did you upload to the romeo ? the modifed one you describe in you ping sensor post ?
Do you like your new romeo board ?
Thanks for your help, you're doing a great job.

cybertiti, 2015-04-06:
You were right, moving away from pin 0 & 1 did the trick.
Ping sensors are working perfectly now.
But I still have an issue with IR sensors reporting "null" value only ...

rwaldron, 2015-04-06:
@cybertiti I'm out of town for a few days so I won't be able to test this until I get back—I appreciate your patience :)

rwaldron, 2015-10-08:
Can you update this to use the Proximity class and let us know if the issue remains?

http://johnny-five.io/api/proximity/

rwaldron, 2015-11-05:
@cybertiti I'm going to close this for now—if you want to dig in again, we can reopen :)

closed 2015-11-05 by rwaldron