rwaldron/johnny-five #677: Usb keyboard simulator

henricavalcante, 2015-03-27:
We can simulate a keyboard using an usb shield, firmata and johnny-five?

rwaldron, 2015-03-27:
I'm not sure, I've never tried. Can you provide any links to existing Arduino C examples?

henricavalcante, 2015-03-27:
I'll try tonight and show my experiments here.

Resseguie, 2015-04-02:
@henricavalcante that's essentially what the MakeyMakey does, right? Could you look at their source for inspiration? The arduino code is under the firmware folder here:
https://github.com/sparkfun/makeymakey

Resseguie, 2015-04-02:
Looks like it just uses the Arduino Keyboard library:
http://arduino.cc/en/Reference/MouseKeyboard

henricavalcante, 2015-04-21:
I found this file https://github.com/practicalarduino/VirtualUsbKeyboard/blob/master/VirtualUsbKeyboard.pde that uses this project http://code.rancidbacon.com/ProjectLogArduinoUSB, can I use something like this with johnny five?

rwaldron, 2015-07-11:
Out of scope

closed 2015-07-11 by rwaldron