rwaldron/johnny-five #685: Make: JavaScript Robotics PDF

kukulako, 2015-04-02:
GREETINGS, I bought the book in pdf format. I thought that explained how to program from scratch in johnny five, but was not, someone could tell me where to learn from scratch? . that whenever I find a tutorial discusses javascript DOM ... anyone know?

Resseguie, 2015-04-02:
Just to clarify, are you looking to get started with JavaScript in general, or Johnny-Five specifically? The book assumed basic familiarity in JavaScript. It's really geared towards software developers that want to try their hand at hardware using a language they are already comfortable with.

Resseguie, 2015-04-02:
As suggested in #440 Johnny-Five can be a good project to start flexing your wings in JavaScript once you have some of the very basics down pat. I say jump into the examples folder or start with one of the projects in that book and just start tinkering.

Resseguie, 2015-04-02:
"tell me if a book JavaScript jQuer jon Duckett it a good book for learn javascript and then learn jhoony five"

@kukulako I got your email about book recommendations, but I'm not familiar with that one. I thought I'd bring the question back here in case anyone else wants to chime in with suggestions before we close this issue.

Though I haven't read it, the book you referenced (http://amzn.to/1Hni6QI) looks to be nicely laid out and Chapters 1-4 probably cover the basics of JS you'd need. (You won't need the DOM stuff covered after that for Johnny-Five, but it's good to know.)

For another options, it's hard to go wrong with "JavaScript: The Definitive Guide"
http://amzn.to/1y3O5Wu

And I got a lot of useful information from "JavaScript: The Good Parts"
http://amzn.to/1I84AUm

Keep in mind that you'll also want to get familiar with Node.js which is just server-side JavaScript. That's what Johnny-Five uses. (Thus "nodebots") Here is a long list of resources to get started:
http://stackoverflow.com/questions/2353818/how-do-i-get-started-with-node-js

Resseguie, 2015-04-02:
I'm going to mark this issue as closed, but others can still feel free to chime in with suggestions on books and tutorials.

closed 2015-04-02 by Resseguie