rwaldron/johnny-five #687: Motion: Refactor motion class and consolidate sensors

frxnz, 2015-04-03:
Similar to #631, IR motion sensors (and motion sensors of any other type) could be consolidated into one `Motion` class.

* [Passive Infrared](https://github.com/rwaldron/johnny-five/blob/master/lib/pir.js): Senses an object entering or leaving range
* [GP2Y0D805Z0F](https://github.com/rwaldron/johnny-five/blob/master/lib/ir.js#L22): Senses an object entering or leaving range

```js
new five.Motion({
  device: "GP2Y0D805Z0F"
});

new five.Motion({
  device: "PIR",
  pin: 7
});

// Object enters sensor range
sensor.on("motionstart", callback);

// Object leaves sensor range
sensor.on("motionend", callback);

```

frxnz, 2015-04-03:
Just to reinforce the case for this being `Motion` instead of `IR.Motion`, there are [devices like this](https://www.sparkfun.com/products/12875) that use ultrasonic sensors instead of infrared sensors.

rwaldron, 2015-04-03:
> Just to reinforce the case for this being Motion instead of IR.Motion

Good call :)

> device: "GP2Y0D805Z0F"

`controller: "GP2Y0D805Z0F"`

frxnz, 2015-04-03:
> `controller: "GP2Y0D805Z0F"`

Looking for clarification on this in #688

Resseguie, 2015-04-29:
@frxnz this can close now, right?

frxnz, 2015-04-29:
@Resseguie Nope, `Proximity` closed #631. `Motion` is next on my list.

Resseguie, 2015-04-29:
Got it!

rwaldron, 2015-05-18:
This is done!

closed 2015-05-18 by rwaldron