rwaldron/johnny-five #695: Better organization for breadboards and generated docs

Resseguie, 2015-04-06:
In #693 I found a couple aspects of the examples and generated docs process to be troublesome.

1) The `grunt examples` task updates docs for examples in programs.json but doesn't delete previously generated docs that are no longer referenced. We had a bunch of old outdated .md files in there.
2) To fix above, I just cleared out the docs folder before regenerating, but that required being careful that you don't accidentally delete the breadboard folder that contains all the diagrams.
3) There are also lots of duplicate breadboards (think led, led-blink, etc) that are difficult to maintain because you have to manually copy an updated breadboard to all the associated examples.

I have the following recommendations:

1) Move breadboards to a sub folder under eg and have a grunt task that clears the docs folder and copies breadboards from eg/breadboard to docs/breadboard before generating examples
2) Instead of requiring duplicate files for breadboards, add a breadboards.json that maps an example to a breadboard. If nothing is included for an example in breadboards.json, default to current behavior looking for one with the same name, or none included if that's also not found. This would significantly reduce the number of breadboard files we're maintaining and also make it easier to update them if needed in the future (like I just did to make them all look more consistent).

Example breadboards.json snippet:

``` js
{
  "led.js": "led-13",
  "led-blink.js": "led-13",
  "led-pulse.js": "led-11-pwm",
  "led-fade.js": "led-11-pwm",
  "led-demo-sequence.js": "led-11-pwm",
  "led-slider.js": "led-11-pwm",
  "led-rgb.js": "led-rgb",
  "led-rainbow.js": "led-rgb"
}
```

rwaldron, 2015-04-06:
> Move breadboards to a sub folder under eg and have a grunt task that clears the docs folder and copies breadboards from eg/breadboard to docs/breadboard before generating examples

This will break links to those files :\ (I've used the raw links in slide decks). I think it's safe to just glob for the files you want to delete. 

> Instead of requiring duplicate files for breadboards, add a breadboards.json that maps an example to a breadboard. If nothing is included for an example in breadboards.json, default to current behavior looking for one with the same name, or none included if that's also not found. 

I like this alot. If we change it to: 

```
{
  "led.js": ["led-13-uno", "led-7-spark", "led-whatever-someotherboard"],
}
```

Then we will solve another major problem that's been haunting me: how to **easily** include multiple diagrams for an example. Eventually I want diagrams for every single supported platform :)

Resseguie, 2015-04-06:
> This will break links to those files

But what if we still checked in the docs folder as we do today, it would just be fully generated from the originals under eg/breadboard.

Relying on non-generated files in docs and docs/breadboards is dangerous because I likely just deleted any such non-generated files in #693 for example. Do I need to put those back? When doing cleanups, how do we know which are safe to delete?

I don't feel super strongly about this, so feel free to ignore. I just know that we had lots of outdated stuff in there with potentially non-working examples (or at least not taking advantage of updated Johnny-Five versions).

I like the addition of multiple breadboards to my original idea. We should definitely add that if we do this.

@rwaldron Related to this, what was the reason that titles.json is separate and not just part of programs.json? Keeping the two in sync is a bit of a pain, and will only get worse if we add breadboards.json. Could both those not  be included in programs.json instead?

``` js
    "topic": "Board",
    "classes": ["Board"],
    "files": [{
      "board.js" : {
        title: "Board",
        breadboards: ["led-13"]
      },
      "board-with-port.js": {
        title: "Board - Specify port",
        breadboards: ["led-13"]
      },
//...
    }]
  },
```

Resseguie, 2015-04-07:
Fixed by #699

closed 2015-04-07 by Resseguie