rwaldron/johnny-five #698: Remove gripper component

Resseguie, 2015-04-06:
Deemed too specific.

rwaldron, 2015-04-06:
Agreed.

rwaldron, 2015-04-06:
OR...

If we can refactor this into a controller/device model and add support for several different types of gripper components, then we can keep it.

Resseguie, 2015-05-02:
@rwaldron So I now vote that we plan to refactor this later to the controller/device model but not let it hold up 0.9.0. You good with removing that label from this issue?

rwaldron, 2015-05-02:
Works for me

dtex, 2018-01-08:
So gripper is just a scalar applied to a servo and I don't believe it warrants having its own class. I'm going to argue that we move this out to a Johnny-Five module and deprecate the gripper class.

dtex, 2018-01-11:
Hi @Resseguie ,

I caught a bit of the CodeMash drone feed... Flying those things looks hard!

Johnny-Five contributors and maintainers are loathe to close issues where someone has a need. We don't want anyone to ever feel that we don't care, but we just haven't gotten to this yet. Rather than leave it languishing as an open issue we have created a [Requested Features](../wiki/Requested-Features) page and added your request for a Gripper module.

This is like the 4th request of yours that I've closed in the past two days. Hopefully will be able to start landing some of these features soon.

closed 2018-01-11 by dtex