rwaldron/johnny-five #700: breadboards.json: Update breadboards.json format to use an array of objects

rwaldron, 2015-04-07:
From: https://github.com/rwaldron/johnny-five/pull/699#issuecomment-90308662

> I could change the simple breadboards.json format to use an object instead of an array, then you could assign a title to each diagram and have a header that visually separates them if multiples are given. 

Slight modification: Update breadboards.json format to use an *array* of *objects*

Example: 

```
  "led.js" : [
    { breadboard: "led-13", title: "", description: "" }, 
    { breadboard: "led-resistor", title: "", description: "" }, 
    { breadboard: "led-13-galileo", title: "", description: "" }, 
    { breadboard: "led-13-edison-arduinoboard", title: "", description: "" }, 
    { breadboard: "led-13-mega", title: "", description: "" }
  ]
}
```

rwaldron, 2015-04-07:
Thanks for jumping on this :)

Resseguie, 2015-04-07:
I was already contemplating doing exactly that. Also helps solve some of the markdown formatting I wasn't happy with.

PR #703 (2015-04-08): Combines example .json files
Fixes #702 to combine programs.json, titles.json, and breadboards.json (with tweaks to proposed format for programs.json)
Fixes #700 to add breadboards as objects instead of array
Went ahead and fixed #701 while I was at it, updating the breadboards listings to get rid of as many duplicate diagrams as I noticed. 

Doesn't add wiki links yet in the generated docs. (Can make that separate issue. Requires adding per-example class definitions to programs.json)
Still could use help adding more descriptions to all examples in programs.json.
closed 2015-04-08 by rwaldron