rwaldron/johnny-five #704: neurotic diatribe about jsdoc, types, and optional parameters

boneskull, 2015-04-08:
I see a lot of this:

```js
/**
 * Does a thing
 * @param {Number} foo It's a foo
 * @param {String} bar It's a bar
 */
function Baz(foo, bar) {
  // ...
}
```

To my knowledge and experience with JSDoc3, this is at least somewhat incorrect.  A `Number` is an object instantiated with the `new` keyword; likewise with `String`.

```js
/**
 * Does a thing
 * @param {number} foo It's a foo
 * @param {string} bar It's a bar
 */
function Baz(foo, bar) {
  // ...
}
```

I think the above is what we want here.  It may not make any practical difference, but using `Number` introduces ambiguity about the parameter's type, while `number` does not.

Something of type `Array` should still be an `Array` because either way you look at it, it's an object.

There are also many places where optional parameters are not properly described or are simply incorrect (and isn't `addr` actually the address of the *device*?  Anyway...).  For example:

```js
   /**
     * led or setLed Set the status of a single Led.
     *
     * @param {Number} addr Address of Led
     * @param {Number} row Row number of Led (0-7)
     * @param {Number} column Column number of Led (0-7)
     * @param {Boolean} state [ true: on, false: off ] [ 1, 0 ] or an LedControl color code
     *
     */
    led: function(addr, row, col, state) {
      // ..
    }
```

(As an aside, isn't it more common for the optional parameter(s) to come *after* the required ones?)

Because of this, consumers of j5 running JSDoc-aware IDEs will notice warnings when they attempt to use this function w/o the optional `addr` parameter.  Perhaps this:

```js
   /**
     * Set the status of a single Led.
     *
     * @param {number} [addr] Address of Led
     * @param {number} row Row number of Led (0-7)
     * @param {number} col Column number of Led (0-7)
     * @param {(number|boolean)} state [ true: on, false: off ] [ 1, 0 ] or an LedControl color code
     *
     */
    led: function(addr, row, col, state) {
      // ..
    }
```

*But*, as extra ammo for the "optional-parameters-last" argument, it's ambivalent whether or not passing three parameters means you forgot the fourth one, because the types are all the same.  The solution then, is to make the fourth parameter optional as well.  But if *that* parameter is optional, then the third parameter will need to accept a boolean value:

```js
   /**
     * Set the status of a single Led.
     *
     * @param {number} [addr] Address of Led
     * @param {number} row Row number of Led (0-7)
     * @param {(number|boolean)} col Column number of Led (0-7) or state if addr is omitted
     * @param {(number|boolean)} [state] [ true: on, false: off ] [ 1, 0 ] or an LedControl color code
     *
     */
    led: function(addr, row, col, state) {
      // ..
    }
```

Of course, `state` is not really optional, because you need no less than three parameters for this particular function (you could *make* it optional, and do something fancy like toggle the value if `state` is omitted and simply switch it "off" if the current value is a color code; of course, your inline documentation shouldn't be influencing your API).

So, if we *were* to change the function signature (necessitating a major release):

```js
   /**
     * Set the status of a single Led.
     *
     * @param {number} row Row number of Led (0-7)
     * @param {number} col Column number of Led (0-7)
     * @param {(number|boolean)} state [ true: on, false: off ] [ 1, 0 ] or an LedControl color code
     * @param {number} [addr] Address of Led
     */
    led: function(row, col, state, addr) {
      // ..
    }
```

No ambiguity & easy to document.

Also, the wiki seems to be missing some information.  I'm not sure why the API documentation isn't pulled directly from the code?  Seems like it'd save time in the long run.  There's a handful of tools out there to convert JSDoc to Markdown for the wiki, or failing that, a JSDoc template for publishing directly to HTML.

rwaldron, 2015-04-08:
> To my knowledge and experience with JSDoc3, this is at least somewhat incorrect. A Number is an object instantiated with the new keyword; likewise with String

Yep.

> There are also many places where optional parameters are not properly described or are simply incorrect (and isn't addr actually the address of the device? Anyway...). For example:

Yep.

------

LedControl was originally written as a port of LedControl.cpp, which was an awesome contribution, but 3 years later I wish this wasn't the case. 

> (As an aside, isn't it more common for the optional parameter(s) to come after the required ones?)

W/r to `addr`: it's not really "optional", I just hate the idea of making users write code that doesn't have an obvious explanation, eg. the address number of a device in a collection of devices is strange if you only have 1 of these devices in your project. 

```js
let x = 0, y = 0, state = 1;

matrix.led(0, x, y, state); // what the hell is that first zero for?
```

Basically, I was trying to optimize for the most common case: one device.

> led: function(row, col, state, addr) {

This prevents creating a function that has the `addr` argument curried: https://github.com/rwaldron/johnny-five/blob/master/lib/ledcontrol.js#L310-L325

mMerlin, 2015-05-07:
FYI the link provided (presumably) about the curried argument is dead.  A recent restructure moved that to an led (sub) folder.  Looks like https://github.com/rwaldron/johnny-five/blob/master/lib/led/ledcontrol.js#L312-L327 matches now.  Even without moving files, line number ranges are prone to break.  Maybe https://github.com/rwaldron/johnny-five/blob/affa20e9d6ae2c38074eaa6a7e65a5ba88a4ebfc/lib/led/ledcontrol.js#L310-L325 would be a better choice, since that should ?never? break.

closed 2016-05-25 by reconbot