rwaldron/johnny-five #705: Add links to wiki documentation from generated examples

Resseguie, 2015-04-08:
Make it easier to find the official API documentation from the example pages.

Probably need to add a "classes" property to each example in programs.json so we'll know which page(s) need to be linked.

Resseguie, 2015-04-08:
@lyzadanger You might be interested in tracking or helping on this. I didn't get to it with my PR to combine the various example .json files.

The new programs.json format looks like:

``` js
  {
    "topic": "LED",
    "classes": ["Led"],
    "examples": [
      {
        "file": "led.js",
        "title": "LED",
        "description": "Basic LED example",
        "breadboards": [
          {"name": "led-13", "title": "LED on pin 13 (Arduino UNO)", "description": "Basic example with LED inserted directly into pin 13"},
          {"name": "led-resistor", "title": "LED on pin 13 with breadboard and resistor (Arduino UNO)", "description": "Basic example with LED on a breadboard and demonstrating use of a resistor"},
          {"name": "led-13-mega", "title": "LED on pin 13 (Arduino Mega)", "description": "Basic example with LED inserted directly into pin 13"}
        ],
        "images": [
          {"file": "led.jpg", "title": "LED on pin 13 (Arduino UNO)", "description": "Basic example with LED inserted directly into pin 13"}
        ]
      },
      // ...
    ]
  }
```

You could probably add:

``` js
"classes": ["Led", ...],
```

to each example entry, and then iterate over those in the Gruntfile to produce the wiki links.

lyzadanger, 2015-04-08:
@Resseguie I'd be happy to take this, with the caveat that I'm traveling this week. What's the urgency level, you think? It may be some days before I can do it.

Resseguie, 2015-04-08:
No urgency on my part, it's just something that's come up several times in relation to not being easy to find related API docs.

Resseguie, 2015-04-08:
Shall I assign you @lyzadanger?

lyzadanger, 2015-04-08:
@Resseguie Certainly.

rwaldron, 2015-04-08:
Note that we'll have to write a replacement routine for johnny-five.org's build system

Resseguie, 2015-04-08:
Is that what the  stuff is for in the templates?

If these links go inside those existing start/end blocks, will it work as-is?

Or do you want a replacement that swaps them out for website links to the same API docs?

rwaldron, 2015-04-08:
> Is that what the  stuff is for in the templates?

Probably. That's just used to demarcate sections in the source markdown content for removal

> If these links go inside those existing start/end blocks, will it work as-is?

Yes.

> Or do you want a replacement that swaps them out for website links to the same API docs?

If the classes used in a given example are listed in the programs.json entry, then the website can remove the repo links and make it's own links from that data.

rwaldron, 2015-10-08:
Docs stuff is all in johnny-five.io now

closed 2015-10-08 by rwaldron