rwaldron/johnny-five #715: Document `Led#animate()`

scottgonzalez, 2015-04-09:
This method isn't currently documented, but it seems like it's probably public API. If this *is* intended to be public API, I can write up the documentation. I'm also wondering if the API should be a little more user-friendly:

* Make `duration` optional, defaulting to `1000`, just like `.pulse()` and `.fade()`.
* Optional `complete` callback, also like `.pulse()` and `.fade()`.

There's also an existing TODO in the code to give the `step` callback a default value. Perhaps just make the default increase the brightness to `0xff`? I'm not sure that having a default for `step` really makes sense though.

scottgonzalez, 2015-04-10:
Just adding a note for myself to check the callback implementation for all the animation methods. If `complete` becomes optional, we can simplify some code in methods like `.fade()`.

rwaldron, 2015-04-10:
> This method isn't currently documented, but it seems like it's probably public API. 

Yes

> If this is intended to be public API, I can write up the documentation

That would be awesome :)

> Make duration optional, defaulting to 1000, just like .pulse() and .fade().

Yes

> Optional complete callback, also like .pulse() and .fade().

Yes

> There's also an existing TODO in the code to give the step callback a default value. Perhaps just make the default increase the brightness to 0xff? I'm not sure that having a default for step really makes sense though.

I haven't given it much thought, so I will defer to @Resseguie (and whoever else has valuable input)

Resseguie, 2015-04-10:
Do we know of any case where .animate() is used directly as part of the public API?

rwaldron, 2015-04-10:
> Do we know of any case where .animate() 

No, but it's exposed on `Led` instances and safe enough for users that want to roll their own goodies. No harm, right?

Resseguie, 2015-04-10:
No, it's just never been intuitive to me the relationship between delta/step/etc at a cursory glance. But because I'm slow doesn't mean it shouldn't be there. ;)

scottgonzalez, 2015-04-10:
Maybe some docs will clear that up :wink:

Resseguie, 2015-04-15:
Re: switching to the Animate implementation instead of the separate one in led.js: #730

dtex, 2015-07-20:
Since this issue was originally opened we've changed how animate works and even removed this method from led. Do we want to add it back? If not we can close this.

closed 2015-10-08 by rwaldron