rwaldron/johnny-five #716: Led animation callbacks

scottgonzalez, 2015-04-09:
`.pulse()` takes a callback, which is invoked on every "beat" of the animation, but `.strobe()` doesn't take a callback. Is this intentional?

Note: The callbacks for `.pulse()` and `.fade()` are not documented. Are these intended to be public API? If so, I can document them.

Resseguie, 2015-04-10:
@rwaldron may have a different opinion, but I don't think a callback is useful for `.pulse()` or `.strobe()` since they repeat until `.stop()` is called. I'd be ok with taking it out of `.pulse()`

Callback on `.fade()` should be documented and match `.fadeIn()` and `.fadeOut()` (which are also missing callback docs it appears).

scottgonzalez, 2015-04-10:
I thought the callback for `.pulse()` was odd too, since it's called repeatedly. The only use case that came to mind was trying to pulse something X times and using the callback to track the count, then calling `.stop()`. Removing it seems fine, assuming nobody's aware of people using the callback.

rwaldron, 2015-04-10:
s/strobe/blink/

`pulse` is handled by `animate` because it has more than just "on" and "off" states; because `animate` accepts a callback, `pulse` accepts a callback (to pass through). 

There is no reason why `blink` **couldn't** have a callback that's invoked on each interval as well; however I can imagine a scenario where some code is running on a lower resource platform and that callback prevents the interval from staying as close to the specified rate as possible. It could even be something as seemingly benign as toggle another LED, you can imagine the IO causing an unwanted delay. 

All of that aside, it might be more useful to allow the callback and give the end programmer more mechanisms for expressing their program. If the program experiences a millisecond or two of drift as a result of the callback's operation, then the programmer can adjust the interval accordingly.

Resseguie, 2015-04-10:
Except there's a bug in the pulse implementation so that the callback is only called the first time:
https://github.com/rwaldron/johnny-five/blob/master/lib/led.js#L381-L386

Fix in #720

scottgonzalez, 2015-04-14:
Updated the wiki.

closed 2015-04-14 by scottgonzalez